khoih-prog/WebSockets2_Generic #6: Ethernet and wifi in the same board

FalsinSoft, 2020-12-04:
Hi

I'm using the arduino mkr wifi 1010 with the ethernet shield (W5500) because, in case of no wifi available, ethernet must be the alternative for connect to the server. I wanted to use your library for communicate using both solutions but it seem is not possible. To be more precise the problem is all the library code is all inside the .hpp header (why?) and is not possible to include the same header in two different .cpp (one configured for wifi and other for ethernet). Including in both source files generate a linker error caused by the conflict of the double function body compiled in the same project. Do you know if exist a workaround for have both devices wifi and ethernet working in the same firmware?
Thank you

khoih-prog, 2020-12-05:
Please first read the [**How to create a Minimal, Reproducible Example**](https://stackoverflow.com/help/minimal-reproducible-example)

> When asking a question, people will be better able to provide help if you provide code that they can easily understand and use to reproduce the problem. This is referred to by community members as creating a minimal, reproducible example (reprex), a minimal, complete and verifiable example (mcve), or a minimal, workable example (mwe). Regardless of how it's communicated to you, it boils down to ensuring your code that reproduces the problem follows the following guidelines:
> 
> Your code examples should be…
> 
> …Minimal – Use as little code as possible that still produces the same problem
> …Complete – Provide all parts someone else needs to reproduce your problem in the question itself
> …Reproducible – Test the code you're about to provide to make sure it reproduces the problem
> The rest of this help article provides guidance on these aspects of writing a minimal, reproducible example.

If you need to pull out the complex magic, you need to master certain tricks of the trade.
That means if you'd like to create the complex use-cases, you must possess **certain level of knowledge to solve the underlying problems**, if arise. **Not just sitting and blaming.**

The simplest way you can avoid the issue is to adapt the solution as in [**HOWTO Fix Multiple Definitions Linker Error**](https://github.com/khoih-prog/ESP_WiFiManager#howto-fix-multiple-definitions-linker-error). 
There are many more sophisticated ways to avoid the issue, if you know how the compiling/linking process is working. Then design your program correctly.

Without the **Minimal, Reproducible Example**, I'll close the issue within several days.

FalsinSoft, 2020-12-05:
I have some open source projects here too and I know some developers at the first problem found open an issue and wait for a solution without put any effort in try a solution by theirself but I ensure you is not my case. Before open this issue I tried all the possible solutions that came to my mind. Probably is true that I don't have enough experience in this kind of problems but I have very rarely found header files that contained all the code without the related .cpp file. I don't understand why but I guess there was a reason for this choice. I had also already found the [link you showed me](https://github.com/khoih-prog/ESP_WiFiManager#howto-fix-multiple-definitions-linker-error) but, honestly, I didn't understand what I should do to solve the problem. Anyway in attach an example (Visual Studio Code with PlaftormIO) project showing the issue. Basically there are two .cpp files both importing "WebSockets2_Generic.h" but configured to work one with ethernet and the other with wifi. This simply code generate the linker error.

[TestEthernetWifi.zip](https://github.com/khoih-prog/WebSockets2_Generic/files/5646864/TestEthernetWifi.zip)

Thank you for your help

khoih-prog, 2020-12-08:
Hi,
Nice that you've spent times to study the code and possible solution. Even though it's not working now, but I'm sure the experience will enrich your knowledge and help you in your whole professional life.

"Anything didn't kill you will make you much better", as they always say.

I'm sorry that the library, by design, now doesn't support the **double-dipping use-case (WiFi + Ethernet)**, no matter the h-only or h/cpp way.

I'll consider modify the library to support this so-called **multiple-dipping** use-cases, in the future. But the time-line is not determined.

In the mean time, if you find out any solution, please share to help other users here.

Good Luck,

closed 2020-12-08 by khoih-prog