khoih-prog/WebSockets2_Generic #7: Fast Websocket sending causes stuttering?

polishdude20, 2020-12-05:
I'm using an Arduino Nano 33 iot with the WifiNina client example. I'm sending over data every 16ms and the data all gets to where it needs to go although it stutters. It's as if I'm receiving the data in chunks rather than quickly every 16ms. Do you guys know of some reason this might occur? I'm trying to send string data which looks like "123,145,156,674" per message.

khoih-prog, 2020-12-05:
Thanks for using the library.

Can you provide more information so that the issue you're experiencing can be duplicated, identified and debugged, if exists.

As you know, the process, even though seems simple, requires the **perfect cooperation of every underlying layers**, such as

1. Nano-33-IoT core and hardware
2. WiFiNINA_Generic/WiFiNINA library
3. This WebSockets2 library
4. Your home network: routers, settings, etc.
5. The WebSockets Server you're connecting to
6. many more

It's advisable that you isolate the issue to be related to 1., 2. and/or 3.by using another board, such as **ESP8266/ESP32**. If the problem persists, then you'll know it's possible 3-6. If this is the case, it's better that you post on the issue on [**ArduinoWebsockets Issue**](https://github.com/gilmaimon/ArduinoWebsockets/issues) as it's ESP32/ESP8266 related.

You can also use this independent [**SimpleWebSocket example**](https://github.com/khoih-prog/WiFiWebServer/tree/master/examples/HTTPClient/SimpleWebSocket) of [**WiFiWebServer Library**](https://github.com/khoih-prog/WiFiWebServer) to verify if the issue is related to this WebSockets2 library.

**You have to post the whole sketch, so that other people can duplicate the issue**. Otherwise, nobody can have psychic power to guess what's wrong or waste time to recreate, and the issue will be closed.

polishdude20, 2020-12-05:
Hello, I'm using the example sketch but I've modified it to send a string message every 16ms.

I've tried using it on my ESP32 and it does not have the same problem. The ESP32 works perfectly fine and the data comes in smoothly and quickly. So this leads me to believe it's:
a) not my server
b) not my internet connection
c) possibly not the library? Maybe there's a problem all Arduino Nano 33 IOT's have that causes this?

I've tried using the WifiWebServer library example you suggested on the Nano 33 iot but it still has the same problem.

Here's some of the code I used from that example. I modified it to send data every 16ms.

```
void setup()
{
  Serial.begin(115200);
  while (!Serial);

  Serial.print("\nStarting SimpleWebSocket on " + String(BOARD_NAME));
  Serial.println(" with " + String(SHIELD_TYPE));

  // check for the presence of the shield
#if USE_WIFI_NINA
  if (WiFi.status() == WL_NO_MODULE)
#else
  if (WiFi.status() == WL_NO_SHIELD)
#endif
  {
    Serial.println(F("WiFi shield not present"));
    // don't continue
    while (true);
  }

#if USE_WIFI_NINA
  String fv = WiFi.firmwareVersion();
  if (fv < WIFI_FIRMWARE_LATEST_VERSION)
  {
    Serial.println(F("Please upgrade the firmware"));
  }
#endif

  // attempt to connect to WiFi network
  while ( status != WL_CONNECTED)
  {
    Serial.print(F("Connecting to SSID: "));
    Serial.println(ssid);
    // Connect to WPA/WPA2 network
    status = WiFi.begin(ssid, pass);
  }

  // you're connected now, so print out the data
  printWifiStatus();
  wsClient.begin("/capstone/ws?clientId=1");
}

void loop()
{
  

  while (wsClient.connected()) 
  {
    // send a hello #
    wsClient.beginMessage(TYPE_TEXT);
    wsClient.print("1,234,562,456,777");
    wsClient.endMessage();

 
    delay(16);
  }

  Serial.println("disconnected");
}
```

khoih-prog, 2020-12-05:
As I suspect, I now think this is the issue is **deeper** inside the **WiFiNINA library and/or Nano-33-IoT.**

Can you make a simple sketch using WiFiNINA to demonstrate the issue, then post on [**WiFiNINA Issue**](https://github.com/arduino-libraries/WiFiNINA/issues).

Also search there to see if there is similar issues/fixes.

Hope you find out a solution. Please update then.

I'm closing the issue now because it's not something related to this library.

closed 2020-12-05 by khoih-prog