khoih-prog/WebSockets2_Generic #8: Is it compatible with MRK GSM 1400?

fechidal89, 2020-12-08:
Arduino MKR GSM 1400
SAMD core version 1.8.9
Arduino IDE 1.8.13
MacOS 10.13.6

I was wondering whether I could connect through WebSocket with this library and the MRK GSM 1400 board?

I built the project and ran it with a dummy configuration but I couldn't pass step 1 when it tries to connect.

Output:
01:40:17.153 -> Starting SAMD-Ethernet-Minimal-Client on SAMD MKRGSM1400
01:40:17.153 -> Ethernet using W5x00 and Ethernet Library
01:40:17.153 -> WebSockets2_Generic v1.1.0
01:40:17.153 -> [WS] WebsocketsClient::connect: step 1

Thanks!

khoih-prog, 2020-12-08:
The SAMD MKRGSM1400 is currently supported using Ethernet or WiFiNINA shield.
Are you using **W5x00 Ethernet** shield and connect correctly (The SPI bus, especially CS/SS to pin 10)? 
If not, the system will hang waiting for connection as you're experiencing.

Check and recheck the Ethernet shield wiring and also if it's working by using another board.

closed 2020-12-08 by khoih-prog

fechidal89, 2020-12-08:
I'm connecting through GSM, no by Ethernet.  I'm using something like this:

```
#include <MKRGSM.h>
#include <WebSockets2_Generic.h>

WebsocketsClient client;
GSM gsmAccess(true);  // pass true as param to enable debug mode
GPRS gprs;

void onMessageCallback(WebsocketsMessage message)
{
  Serial.print("Got Message: ");
  Serial.println(message.data());
}

void setup() {

  bool connected = false;

  while (!connected) {
    if ((gsmAccess.begin(PINNUMBER) == GSM_READY) &&
        (gprs.attachGPRS(GPRS_APN, GPRS_LOGIN, GPRS_PASSWORD) == GPRS_READY)) {
      connected = true;
    } else {
      Serial.println("Not connected");
      delay(1000);
    }
  }

  connected = client.connect("ws://192.168.1.121",  80, "/");

  if (connected) {
    Serial.println("Connected...");
  } else {
    Serial.println("NOT Connected...");
  }
   
  client.onMessage(onMessageCallback);
}

void loop() {
  // let the websockets client check for incoming messages
  if (client.available()) {
    client.poll();
  }
}
```

Maybe I c&p your example and you saw that I'm using Ethernet but no, I need to connect through GSM. I test the data making an HTTP request and It works, but when I try to use ws It doesn't work. I test the server-side (ws - nodejs) with the google chrome extension and It works too.

Do I need to set some headers?
Is it possible to use your library with GSM/LTE?

Thanks!

khoih-prog, 2020-12-08:
The GSM/GPRS is not currently supported. I'll consider to add the support in the future if there are many more requests.