khoih-prog/WebSockets2_Generic #9: Compile example failed

Erhannis, 2020-12-11:
I'm trying to compile the example Esp8266-Server.  Arduino IDE 1.8.5.  I get compilation errors.  Here's the log:

```
/home/erhannis/arduino-1.8.5/arduino-builder -dump-prefs -logger=machine -hardware /home/erhannis/arduino-1.8.5/hardware -hardware /home/erhannis/.arduino15/packages -tools /home/erhannis/arduino-1.8.5/tools-builder -tools /home/erhannis/arduino-1.8.5/hardware/tools/avr -tools /home/erhannis/.arduino15/packages -built-in-libraries /home/erhannis/arduino-1.8.5/libraries -libraries /home/erhannis/Arduino/libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=qio,FlashSize=512K0,led=2,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=115200 -ide-version=10805 -build-path /tmp/arduino_build_250835 -warnings=none -build-cache /tmp/arduino_cache_89380 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=/home/erhannis/.arduino15/packages/esp8266/tools/esptool/0.4.13 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/home/erhannis/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=/home/erhannis/.arduino15/packages/esp8266/tools/mkspiffs/0.2.0 -verbose /home/erhannis/clones/WebSockets2_Generic/examples/ESP8266/Esp8266-Server/Esp8266-Server.ino
/home/erhannis/arduino-1.8.5/arduino-builder -compile -logger=machine -hardware /home/erhannis/arduino-1.8.5/hardware -hardware /home/erhannis/.arduino15/packages -tools /home/erhannis/arduino-1.8.5/tools-builder -tools /home/erhannis/arduino-1.8.5/hardware/tools/avr -tools /home/erhannis/.arduino15/packages -built-in-libraries /home/erhannis/arduino-1.8.5/libraries -libraries /home/erhannis/Arduino/libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=qio,FlashSize=512K0,led=2,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=115200 -ide-version=10805 -build-path /tmp/arduino_build_250835 -warnings=none -build-cache /tmp/arduino_cache_89380 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=/home/erhannis/.arduino15/packages/esp8266/tools/esptool/0.4.13 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/home/erhannis/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=/home/erhannis/.arduino15/packages/esp8266/tools/mkspiffs/0.2.0 -verbose /home/erhannis/clones/WebSockets2_Generic/examples/ESP8266/Esp8266-Server/Esp8266-Server.ino
Using board 'generic' from platform in folder: /home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1
Using core 'esp8266' from platform in folder: /home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1
Build options changed, rebuilding all
Detecting libraries used...
"/home/erhannis/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/include" "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/lwip2/include" "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_250835/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=2  -DESP8266 "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266" "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/variants/generic" "/tmp/arduino_build_250835/sketch/Esp8266-Server.ino.cpp" -o "/dev/null"
"/home/erhannis/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/include" "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/lwip2/include" "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_250835/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=2  -DESP8266 "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266" "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/variants/generic" "-I/home/erhannis/Arduino/libraries/WebSockets2_Generic/src" "/tmp/arduino_build_250835/sketch/Esp8266-Server.ino.cpp" -o "/dev/null"
"/home/erhannis/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/include" "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/lwip2/include" "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_250835/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=2  -DESP8266 "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266" "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/variants/generic" "-I/home/erhannis/Arduino/libraries/WebSockets2_Generic/src" "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src" "/tmp/arduino_build_250835/sketch/Esp8266-Server.ino.cpp" -o "/dev/null"
"/home/erhannis/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/include" "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/lwip2/include" "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_250835/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=2  -DESP8266 "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266" "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/variants/generic" "-I/home/erhannis/Arduino/libraries/WebSockets2_Generic/src" "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src" "/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp" -o "/dev/null"
"/home/erhannis/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/include" "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/lwip2/include" "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_250835/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=2  -DESP8266 "-I/home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266" "-I/home/erhannis/.arduino15/packages/esp8266/hardware [... cut ...]
/home/erhannis/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic_Client.hpp:977:11: error: 'class websockets2_generic::WebsocketsClient' has no member named '_optional_ssl_known_key'
     this->_optional_ssl_known_key = pk;
           ^
Using library WebSockets2_Generic at version 1.1.0 in folder: /home/erhannis/Arduino/libraries/WebSockets2_Generic 
Using library ESP8266WiFi at version 1.0 in folder: /home/erhannis/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi 
exit status 1
Error compiling for board Generic ESP8266 Module.
```

khoih-prog, 2020-12-11:
You're using too out-of-date Arduino IDE and ESP8266 core version

You can try to update

1. `Arduino IDE 1.8.5.` into latest `Arduino IDE 1.8.13.`
2. `ESP8266 core 2.4.1` to `ESP8266 core 2.7.4`

Then recompile.

---

**Compiled OK**

[image]

---

**Prerequisites**

[image]

closed 2020-12-11 by khoih-prog

Erhannis, 2020-12-15:
Oh, you're right - sorry!  Thanks!