khoih-prog/WebSockets2_Generic #10: not compiling on Arduino DUE with WIZnet W5500-Ethernet shield with SD-slot, IDE: 1.8.14

PeterDuesentrieb, 2020-12-14:
Hi! 
could u help me? Dont know what Im doing wrong. Example: SAMDUE-Ethernet_ServerAllFunctionsDemo.
It s the same with SAMDUE-Ethernet-Server.

I had succes with (WebSockets-Lib)<WebSocketsServer.h> together with <Ethernet.h> but the problem is, that the Websocketsserver.cpp filled the Memory with one and the same client during first connection until he cleaned up everyone because >4 Clients

my shield:
https://www.mouser.de/ProductDetail/WIZnet/W5500-Ethernet-Shield?qs=k5OWtXsTJapDzGtdLOlnnw==

C:\Users\myName\AppData\Local\Temp\arduino_modified_sketch_842186/SAMDUE-Ethernet_ServerAllFunctionsDemo.ino:143: undefined reference to `std::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()'
collect2.exe: error: ld returned 1 exit status
Using library EthernetLarge in folder: C:\Program Files (x86)\arduino-nightly\libraries\EthernetLarge (legacy)
Using library WebSockets2_Generic at version 1.0.7 in folder: C:\Users\myName\Documents\Arduino\libraries\WebSockets2_Generic 
Using library EthernetWebServer at version 1.1.0 in folder: C:\Users\myName\Documents\Arduino\libraries\EthernetWebServer 
Using library Functional-Vlpp at version 1.0.1 in folder: C:\Users\myName\Documents\Arduino\libraries\Functional-Vlpp 
Using library SPI at version 1.0 in folder: C:\Users\myName\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\libraries\SPI 
Using library Ethernet at version 2.0.0 in folder: C:\Users\myName\Documents\Arduino\libraries\Ethernet 
exit status 1
Error compiling for board Arduino Due (Programming Port).

my defines.h:

#ifndef defines_h
#define defines_h

#if ( defined(ARDUINO_SAM_DUE) || defined(__SAM3X8E__) )
  #if defined(WEBSOCKETS_ETHERNET_USE_SAMDUE)
    #undef WEBSOCKETS_ETHERNET_USE_SAMDUE
  #endif
  #define WEBSOCKETS_USE_ETHERNET             true
  #define WEBSOCKETS_ETHERNET_USE_SAMDUE      true
#else
  #error This code is intended to run only on the SAM DUE boards ! Please check your Tools->Board setting.
#endif

#if defined(WEBSOCKETS_ETHERNET_USE_SAMDUE)

// For SAM DUE
#if defined(ARDUINO_SAM_DUE)
#define BOARD_TYPE      "SAM DUE"
#elif defined(__SAM3X8E__)
#define BOARD_TYPE      "SAM SAM3X8E"
#else
#define BOARD_TYPE      "SAM Unknown"
#endif

#endif

#if defined(ARDUINO_BOARD)
  #define BOARD_NAME    ARDUINO_BOARD
#else
  #ifndef BOARD_NAME
    #define BOARD_NAME    BOARD_TYPE
  #endif
#endif

// Just select one to be true. If all is false, default is Ethernet. 
// If more than one are true, the priority is USE_ETHERNET, USE_ETHERNET2, USE_ETHERNET_LARGE, USE_UIP_ETHERNET
#define USE_ETHERNET                  false
#define USE_ETHERNET2                 false
#define USE_ETHERNET_LARGE            true
#define USE_ETHERNET_ENC              false

#define USE_UIP_ETHERNET              false

#if USE_ETHERNET
  // Also default to Ethernet library
  #include <Ethernet.h>
  #define ETHERNET_TYPE               "W5x00 and Ethernet Library"
#elif USE_ETHERNET2
  #include <Ethernet2.h>
  #define ETHERNET_TYPE               "W5x00 and Ethernet2 Library"
#elif USE_ETHERNET_LARGE
  #include <EthernetLarge.h>
  #define ETHERNET_TYPE               "W5x00 and EthernetLarge Library"
#elif USE_ETHERNET_ENC
  #include <EthernetENC.h>
  #define ETHERNET_TYPE               "ENC28J60 and EthernetENC Library"  
#elif USE_UIP_ETHERNET
  #include <UIPEthernet.h>
  #include <utility/logging.h> 
  #define ETHERNET_TYPE               "ENC28J60 and UIPEthernet Library"
#else
  // Default to Ethernet library
  #include <Ethernet.h>
  #define ETHERNET_TYPE               "W5x00 and Ethernet Library"
#endif

khoih-prog, 2020-12-14:
It seems that you're using not-latest versions of the prerequisite libraries. I suggest you update these following libraries and retry:

1. [`WebSockets2_Generic v1.1.0`](https://github.com/khoih-prog/WebSockets2_Generic/releases/tag/v1.1.0). To install, check [image]
2. [`EthernetWebServer library v1.2.0+`](https://github.com/khoih-prog/EthernetWebServer) if necessary to use Ethernet modules/shields. To install, check [image]
3. **To be able to compile and run on SAM DUE boards**, you have to copy the whole [SAM DUE](Packages_Patches/arduino/hardware/sam/1.6.12) directory into Arduino sam directory (~/.arduino15/packages/arduino/hardware/sam/1.6.12). 

Supposing the Arduino SAM core version is 1.6.12. This file must be copied into the directory:

- `~/.arduino15/packages/arduino/hardware/sam/1.6.12/platform.txt`

Whenever a new version is installed, remember to copy this file into the new version directory. For example, new version is x.yy.zz
This file must be copied into the directory:

- `~/.arduino15/packages/arduino/hardware/sam/x.yy.zz/platform.txt`

It's possible something wrong with your IDE/core/libraries installation. The worst case is that you have to reinstall them.

I've just compile the same sketch with the same settings and still OK

[image]

Good Luck,

closed 2020-12-14 by khoih-prog

PeterDuesentrieb, 2020-12-14:
because I couldnt compile again, I erased all foreign libs which have to do with ethernet and websockets.
I wanted 2 use only these, which r interated in the \documentsArduino\libraries\WebSockets2_Generic\LibraryPatches .
So I cut em out of "LibraryPatches" and pasted it into: C:\Program Files (x86)\arduino-nightly\libraries
but the ide seems not to find the headerfiles. Any idea what could be wrong?
[image]

khoih-prog, 2020-12-14:
I'm afraid **only you can find out** what's very wrong with your installation, and **why the compiler can't find the library `EthernetLarge`.**

**This has nothing to do with this library.** It's better to search for installation help in the General Arduino IDE forum.

I also recognize that you're using the experimental Arduino IDE 1.8.14 (nightly-built) and suggest that you reinstall the official v1.8.13 and see if it's OK.

PeterDuesentrieb, 2020-12-15:
"I'm afraid only you can find out what's very wrong with your install" I think u r right! I deleted arduino nightly, installed Arduino 1.8.13, deleted every old lib I had, downloaded websockets2-generic with Library Manager, downloaded "Arduino SAM Boards (ARM Cortex-M3) 1.6.12" with Boards Manager, and tried to compile. -> failed.
After downloading Websockets2_generic on a different computer, we had the same problem. (both machines Win10, ide 1.8.13)
After that, I exchanged on one computer the "platform.txt" with ur Version, which is in \WebSockets2_Generic\Packages_Patches\arduino\hardware\sam\1.6.12 -> failed
Maybe I missunderstood something.
Was that right? Do I have to do more changes because of ur enumeration 3. To be able to comp...
[image]

PeterDuesentrieb, 2020-12-15:
Now I changed every "#include <EthernetLarge.h>" to 
#include "C:\Users\myName\Documents\Arduino\libraries\WebSockets2_Generic\LibraryPatches\EthernetLarge\src\EthernetLarge.h" (also in DUE_Ethernet_tcp.hpp) -> failed

Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "Arduino Due (Programming Port)"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\myName\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\myName\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\myName\Documents\Arduino\libraries -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=2341_003D -ide-version=10813 -build-path C:\Users\myName\AppData\Local\Temp\arduino_build_140142 -warnings=default -build-cache C:\Users\myName\AppData\Local\Temp\arduino_cache_235242 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\myName\Documents\ArduinoData\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\myName\Documents\ArduinoData\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.bossac.path=C:\Users\myName\Documents\ArduinoData\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=C:\Users\myName\Documents\ArduinoData\packages\arduino\tools\bossac\1.6.1-arduino -verbose C:\Users\myName\AppData\Local\Temp\arduino_modified_sketch_30824\SAMDUE-Ethernet_ServerAllFunctionsDemo.ino

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\myName\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\myName\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\myName\Documents\Arduino\libraries -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=2341_003D -ide-version=10813 -build-path C:\Users\myName\AppData\Local\Temp\arduino_build_140142 -warnings=default -build-cache C:\Users\myName\AppData\Local\Temp\arduino_cache_235242 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\myName\Documents\ArduinoData\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\myName\Documents\ArduinoData\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.bossac.path=C:\Users\myName\Documents\ArduinoData\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=C:\Users\myName\Documents\ArduinoData\packages\arduino\tools\bossac\1.6.1-arduino -verbose C:\Users\myName\AppData\Local\Temp\arduino_modified_sketch_30824\SAMDUE-Ethernet_ServerAllFunctionsDemo.ino

Using board 'arduino_due_x_dbg' from platform in folder: C:\Users\myName\Documents\ArduinoData\packages\arduino\hardware\sam\1.6.12

Using core 'arduino' from platform in folder: C:\Users\myName\Documents\ArduinoData\packages\arduino\hardware\sam\1.6.12

Detecting libraries used...

"C:\\Users\\myName\\Documents\\ArduinoData\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10813 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-IC:\\Users\\myName\\Documents\\ArduinoData\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-IC:\\Users\\myName\\Documents\\ArduinoData\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-IC:\\Users\\myName\\Documents\\ArduinoData\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-IC:\\Users\\myName\\Documents\\ArduinoData\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-IC:\\Users\\myName\\Documents\\ArduinoData\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "C:\\Users\\myName\\AppData\\Local\\Temp\\arduino_build_140142\\sketch\\SAMDUE-Ethernet_ServerAllFunctionsDemo.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for WebSockets2_Generic.h: [WebSockets2_Generic@1.1.0]

ResolveLibrary(WebSockets2_Generic.h)

  -> candidates: [WebSockets2_Generic@1.1.0]

"C:\\Users\\myName\\Documents\\ArduinoData\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions [... cut ...]
C:\Users\myName\AppData\Local\Temp\arduino_modified_sketch_30824/SAMDUE-Ethernet_ServerAllFunctionsDemo.ino:100: undefined reference to `Ethernet'

collect2.exe: error: ld returned 1 exit status

Using library WebSockets2_Generic at version 1.1.0 in folder: C:\Users\myName\Documents\Arduino\libraries\WebSockets2_Generic 

Using library EthernetWebServer at version 1.2.0 in folder: C:\Users\myName\Documents\Arduino\libraries\EthernetWebServer 

Using library Functional-Vlpp at version 1.0.1 in folder: C:\Users\myName\Documents\Arduino\libraries\Functional-Vlpp 

exit status 1

Error compiling for board Arduino Due (Programming Port).

khoih-prog, 2020-12-15:
This post is much better and give the info to know what's happening.

The issue is that the compiler can't find the `EthernetLarge` library

```
undefined reference to `Ethernet'
```

I just can suggest that you check

1. Did you install the `Ethernet` and/or `EthernetLarge` Library ? The `Libraries Patches` just provide 1 file that need to change. You need the whole `Ethernet` and/or `EthernetLarge`  library, not just that 1 file patch.
2. You're not correct in changing every 

```#include <EthernetLarge.h>```
to

```
#include "C:\Users\myName\Documents\Arduino\libraries\WebSockets2_Generic\LibraryPatches\EthernetLarge\src\EthernetLarge.h"
```

Please restore them after installing the `Ethernet` and/or `EthernetLarge` library.

The Arduino IDE is smart enough to find out and use the `Ethernet` and/or `EthernetLarge` provided that you install them correctly

3.  `Ethernet` library can be installed using `Library Manager`

[image]

so I suggest that you change the code to use it. You must see `INSTALLED` mark to know that IDE recognizes the library (see picture).

4.  `EthernetLarge` libary can be installed by download directly from [**EthernetLarge Libary**](https://github.com/OPEnSLab-OSU/EthernetLarge)

5. Another good choice is using `Ethernet2` library which you can install directly with `Library Manager`

[image]

You must see `INSTALLED` mark to know that IDE recognizes the library (see picture)

I normally don't have time to help in these basic steps, but understand that you have tried your best and spent time to research then post the complete info and retry my best to help now. Hopefully OK now.

Good Luck,

khoih-prog, 2020-12-15:
The `EthernetLarge Library` directory includes many files

[image]

PeterDuesentrieb, 2021-01-18:
Hello Khoi!
I wish u a succesful and healthy new year! I followed ur advice, thx a lot! Now I can compile. BUT...

1. after changing the IP-Adress in defines.h (of SAMDUE-ETHERNET_ServerAllFunctionsDemo) and also changing in main-ino row 86, the mozilla sais: "Uncaught ReferenceError: wsServer is not defined"
I think it is because "wsServer" is for Mozilla just Text, not a Value.
So I changed row87 to:

    server.send(200, "text/html; **charset=utf-8**", "<html><head><script>var connection = new WebSocket('**ws://' + window.location.hostname +':8080/**', ['arduino']);connection.onopen = function () {  connection.send('Connect ' + new Date()); }; connection.onerror = function (error) {    console.log('WebSocket Error ', error);}; connection.onmessage = function (e) {  console.log('Server: ', e.data);}; **connection.onclose = function () {connection.send('WebSocket connection closed**'); console.log('WebSocket connection closed');}; function sendRGB() {  var r = parseInt(document.getElementById('r').value).toString(16);  var g = parseInt(document.getElementById('g').value).toString(16);  var b = parseInt(document.getElementById('b').value).toString(16);  if(r.length < 2) { r = '0' + r; }   if(g.length < 2) { g = '0' + g; }   if(b.length < 2) { b = '0' + b; }   var rgb = '#'+r+g+b;    console.log('RGB: ' + rgb); connection.send(rgb); }</script></head><body>LED Control:<br/><br/>R: <input id=\"r\" type=\"range\" min=\"0\" max=\"255\" step=\"1\" oninput=\"sendRGB();\" /><br/>G: <input id=\"g\" type=\"range\" min=\"0\" max=\"255\" step=\"1\" oninput=\"sendRGB();\" /><br/>B: <input id=\"b\" type=\"range\" min=\"0\" max=\"255\" step=\"1\" oninput=\"sendRGB();\" /><br/></body></html>");

+_charset_ just to avoid a warning of Mozilla
+_window.location.hostname_ to avoid typing in myIP a second time.

2. After I thought, its runnging, I saw in the Console of Mozilla and serial output the handshake and 'Connect + Date'. When I change the color code, I can see that in Mozilla console but no Feedback to my serial output. Thats why I added "**connection.onclose**".
[image]
my serial output:
[image]
do u have any idea, why the websocket closes?

khoih-prog, 2021-01-18:
It's good you made good progress and the DUE is working now using the WebSocket.

From your terminal output, I see everything is OK for you now. 

You have to understand that the examples are just very simple examples, just to demonstrate the basic operations of the WebSockets. To receive and use the RGB input you saw on the WebBrowser, you have to write a **Web request handler** for the WebServer code, depending on the script in HTML. There is nothing in the code to handle the RBG info now, that's why you didn't see anything relating to it on Serial terminal. I guess you have to do **further HTML script research** then write code if you need to receive the RGB input.

The example is written to **close the WS connection right after receiving the message from WS Client.** You can modify it as you like. Just remember examples are just **dull examples to show you basic operations and how to use certain basic commands.** Don't expect to use directly in any application without modifications.

You can use this [**SAMDUE-Ethernet-RepeatingClient**](https://github.com/khoih-prog/WebSockets2_Generic/tree/master/examples/Generic/Ethernet/SAMDUE/SAMDUE-Ethernet-RepeatingClient) or similar examples to test the WebSockets Server function of the WS Server you created on DUE.

This is all the library can help you, and **you're on your own now** to modify and use it. I'm afraid I don't have time to help anymore with basic usage, unless it's a bug of the library.

Good Luck,