khoih-prog/WebSockets2_Generic #11: Docs?

Erhannis, 2020-12-15:
Are there any docs?  API?  The examples are helpful, but for instance, I'm not sure what _exactly_ `client.available()` does, nor e.g. how many different callbacks are available.  (Also, it looks like `onEventsCallback` in the README is never used?)  The many-targeted nature of the code makes it hard to search the code for a definitive answer.  It sounds like this project is based on another - is there other documentation elsewhere that is considered canon for this library, too?

khoih-prog, 2020-12-15:
You have to read more carefully the [**README document**](https://github.com/khoih-prog/WebSockets2_Generic/blob/master/README.md) before speaking hastily as you did in your previous post [**Compile example failed**](https://github.com/khoih-prog/WebSockets2_Generic/issues/9)

Certainly, the library requires **certain basic knowledge** to use, even many examples are there to guide the users.
If you don't know what you're doing, I suggest you study the examples to figure out yourself.

I certainly don't time to fulfill all basic requests posted without basic research or knowledge.

Good Luck,

closed 2020-12-15 by khoih-prog

Erhannis, 2020-12-15:
Well, yes, I admit I was hasty in my post the other day, and I apologize for that.  Consider, though, that the README is over 50 pages long, and a lot of it (such as the several pages of changelogs, or the many chip-specific variations) is not useful to a new user.  Also, as I mentioned, there are things the readme doesn't address, such as (again, as before) what `client.available()` actually means, or how many kinds of callback there are.  After further searching, the library from which this one is derived answers some such questions with its [wiki](https://github.com/gilmaimon/TinyWebsockets/wiki) - though it's not always completely accurate for this library.  Still, quite useful, and it might be worth adding a direct link there with a more pointed explanation.  Overall, some things that could help:

1. Add a table of contents
2. Link the TinyWebsockets wiki directly
3. Move some sections into their own files, such as the full changelog, the todo vs done, and perhaps some of the larger chunks of platform-specific notes could be moved into their own appropriately named files.  I'll grant that it can be convenient to have them all in one place, so perhaps a table of contents would be enough to solve the problem.

I've done the first and second items - a table of contents, and linking the wiki directly - if you want to review my pull request?  https://github.com/khoih-prog/WebSockets2_Generic/pull/12

khoih-prog, 2020-12-15:
I'm so sorry for what I said as I thought you were similar to other users by abusing other people's free time to compensate to the time spent for research.

I just hope everybody will be as positive and helpful as you: if seeing something not good or not adequate, please contribute by making Pull-Request. That'll benefit to everybody. 
You know there is so much a single guy can do, know and understand. Maintaining 50+ and writing some more libraries are taking toll and I don't have much time to help every single user everything as I did wish. I have to think about bigger picture that can help many more people.

I'll merge the PR. Thanks so much and more contributions are always appreciated.

Best Regards,

KH

Erhannis, 2020-12-15:
To be fair, I opened the other issue early, before I'd tried all the reasonable things I'd considered, because I was impatient and in a hurry.  I will try to be more careful in the future.  Thanks for your work!