khoih-prog/WebSockets2_Generic #14: Missing stm32_def.h for Teensy 4.1

hendrikjvr, 2021-03-16:
PlatformIO Core, version 5.1.0
Windows 10 Enterprise
Board: Teensy 4.1
Framework: Arduino

Hi, I am trying to build a project in PlatfromIO for the Teensy 4.1 with this library added to platform.ini. It does not compile successfully. I get the following error 8 times before the compilation is terminated:

    ".pio\libdeps\teensy41\STM32Ethernet\src\utility\stm32_eth.h:42:23: fatal error: stm32_def.h: No such file or directory"

Why would there be STM32 files under the teensy41 core?

My platform.ini file looks as follows:

    [env:teensy41]
    platform = teensy
    board = teensy41
    framework = arduino
    lib_deps = 
        https://github.com/vjmuzik/FNET.git
        https://github.com/vjmuzik/NativeEthernet.git
        https://github.com/sstaub/TeensyID.git
        https://github.com/khoih-prog/WebSockets2_Generic.git

Steps to reproduce:
1. Open PIO project for Teensy 4.1 board
2. Add Github link for FNET to platformo.ini file: https://github.com/vjmuzik/FNET.git
3. Add Github link for NativeEthernet to platformio.ini file: https://github.com/vjmuzik/NativeEthernet.git
4. Add Github link for TeensyID to platformio.ini file: https://github.com/sstaub/TeensyID.git
5. Add Github link for WebSockets2_Generic to platformio.ini file: https://github.com/khoih-prog/WebSockets2_Generic.git
6. Build project

hendrikjvr, 2021-03-17:
Thanks for your feedback. I will revert here if issue is resolved.

hendrikjvr, 2021-03-17:
I removed all unnecessary lines in `library.json` for our case and removed all folders under `libdeps\teensy41` that we won't need. Can build now without any errors. We are still waiting for our Teensy 4.1, so will confirm if upload is successful and websocket example works when it arrives.

khoih-prog, 2021-03-22:
I'm closing this issue now. If you have new issue after receiving Teensy 4.1, please post new issue.

closed 2021-03-22 by khoih-prog