khoih-prog/WebSockets2_Generic #15: multiple definition of `websockets2_generic::WebsocketsServer::....'

OOHehir, 2021-04-14:
Hello,

Firstly thanks for the work you've put into the library.

I'm using ver 1.1 of your library in conjunction with Sloeber on Eclipse on a 8266.

I have an issue which seems related to a closed bug report (https://github.com/khoih-prog/WebSockets2_Generic/issues/3) but I haven't been able to solve it with your proposed solution.

I have the following setup:

sensor.ino
```

#include "LwDevice.h"
#include <Arduino.h>

lwDevice lwNode;

setup()

loop()
```

lwDevice.cpp
```

#include "LwDevice.h"
#include <Arduino.h>

int LwDevice::sendSocketJSON(){

//websockets2_generic::WebsocketsClient socket; Compiles ok if initialised here
do socket stuff...

}
```

lwDevice.h

```
#include <WebSockets2_Generic.h>

class LwDevice {
private:
websockets2_generic::WebsocketsClient socket; // lots of multiple def errors when init here

}

```
So when creating the object inside the sendSocketJSON() it compiles correctly, moving it to the class causes multiple definition errors. 
E.g:
/opt/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.7.0/libraries/EEPROM/EEPROM.cpp:108: multiple definition of `EEPROMClass::write(int, unsigned char)'; ./libraries/ESP_EEPROM/src/ESP_EEPROM.cpp.o:/opt/eclipse/arduinoPlugin/libraries/ESP_EEPROM/2.1.1/src/ESP_EEPROM.cpp:296: first defined here
/opt/eclipse/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: ./libraries/EEPROM/EEPROM.cpp.o: in function `EEPROMClass::commit()':
/opt/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.7.0/libraries/EEPROM/EEPROM.cpp:127: multiple definition of `EEPROMClass::commit()'; ./libraries/ESP_EEPROM/src/ESP_EEPROM.cpp.o:/opt/eclipse/arduinoPlugin/libraries/ESP_EEPROM/2.1.1/src/ESP_EEPROM.cpp:339: first defined here
/opt/eclipse/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: ./libraries/EEPROM/EEPROM.cpp.o: in function `EEPROMClass::end()':
/opt/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.7.0/libraries/EEPROM/EEPROM.cpp:81: multiple definition of `EEPROMClass::end()'; ./libraries/ESP_EEPROM/src/ESP_EEPROM.cpp.o:/opt/eclipse/arduinoPlugin/libraries/ESP_EEPROM/2.1.1/src/ESP_EEPROM.cpp:248: first defined here
/opt/eclipse/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: ./libraries/EEPROM/EEPROM.cpp.o:/opt/eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.7.0/libraries/EEPROM/EEPROM.cpp:156: multiple definition of `EEPROM'; ./libraries/ESP_EEPROM/src/ESP_EEPROM.cpp.o:/opt/eclipse/arduinoPlugin/libraries/ESP_EEPROM/2.1.1/src/ESP_EEPROM.cpp:528: first defined here

Any pointers would be gratefully appreciated,

ark02, 2021-04-15:
I am in the same boat as you, just started using lib, same issues
I'm using Platformio 

In another issue, he added the instructions to readme: 
https://github.com/khoih-prog/AsyncHTTPRequest_Generic#howto-fix-multiple-definitions-linker-error

But these were unclear to me... 

What I ended up doing was: 
- Split up ALL websocket functionality into its own header file and cpp file 
- move all ETHwebserver code to its own header and cpp file 
- include the ETHwebserver.h at the bottom of webshocket.h 
- Only call webserver.h in other parts of the code (to pull IP address, etc)

khoih-prog, 2021-04-15:
Thanks for your interests in the library and many thanks to @ark02 for your help.

Without the [**MRE**](https://stackoverflow.com/help/minimal-reproducible-example), I'm afraid nobody can help you, especially with your seemingly complicated project.

It also seems that the error messages are relating to certain **EEPROMClass** you're using, nothing relating to this library.

```
multiple definition of EEPROMClass::commit()';
```

Carefully study solved issue :  [**multiple definition** of `websockets2_generic::WebsocketsServer::~WebsocketsServer()' in websocket2_generic library #3](https://github.com/khoih-prog/WebSockets2_Generic/issues/3) to understand why the issue happens. Then I think you can fix it yourself with your good knowledge to do such a complicated project. Ask yourself a question why many people have been using this library without that issue, then restructure your code.

Another way to ask for help is to post the issue in [**ArduinoWebsockets**](https://github.com/gilmaimon/ArduinoWebsockets) because you're using ESP8266, certainly after you've tried to compile using that library and get the same error.

I'm closing the issue now, and will reopen **whenever you prove that the issue is a bug of this library.**

Good Luck,

closed 2021-04-15 by khoih-prog