khoih-prog/WebSockets2_Generic #16: Secured-Esp32-Client Example not works on NodeMcu32s (Esp32s)

anand-010, 2021-04-15:
Secured-Esp32-Client Example on esp32 shows not connected. I am also tried different working wss links and none of them working.
[image]
I am using arduino 15 IDE, WebSockets2_Generic v1.1.0
checked access point internet connection.
Serial monitor logs
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4

Starting WebSockets Secured-ESP32-Client on ESP32_DEV
WebSockets2_Generic v1.1.0
.....Connected to Wifi, Connecting to WebSockets Server @wss://echo.websocket.org/
[WS] WebsocketsClient::doestStartsWith: str = wss://echo.websocket.org/
[WS] WebsocketsClient::doestStartsWith: prefix = http://
[WS] WebsocketsClient::doestStartsWith: str = wss://echo.websocket.org/
[WS] WebsocketsClient::doestStartsWith: prefix = ws://
[WS] WebsocketsClient::doestStartsWith: str = wss://echo.websocket.org/
[WS] WebsocketsClient::doestStartsWith: prefix = wss://
Not Connected!

```

khoih-prog, 2021-04-15:
As you're using ESP32, it's better you post on [**ArduinoWebsockets issue**](https://github.com/gilmaimon/ArduinoWebsockets) where Gil Maimon is certainly more knowledgeable and experience than me on ESP-related issues., if you are in  a rush.

Did you check if the SSL CA Cert. for `echo.websocket.org` is still valid, or missing from your code?
Did you try to access other **websockets** sites (certainly with corresponding and **updated** CA Certs)?
Which esp32 core are you using (v1.0.4, v1.0.5 or v1.0.6?

I currently don't have time to spend on this issue yet, but will follow up and work on it whenever I'm available.

Good Luck,

khoih-prog, 2021-04-15:
[image]

It's possible `echo.websocket.org` has just used new CA Certs from Mar 15th 2020.

khoih-prog, 2021-04-15:
It is confirmed that new CA Cert is necessary.  Even ESP8266 has the same issue.

For ESP8266, change **fingerprint** in [**Secured-Esp8266-Client** defines.h](https://github.com/khoih-prog/WebSockets2_Generic/blob/master/examples/ESP8266/Secured-Esp8266-Client/defines.h) will make it working again.

```
...
// To update SHA1 fingerprint, use Google Chrome to connect to https://www.websocket.org/echo.html 
// Then "View Site Information" => "Certificate Viewer" => Copy SHA1 fingerprint
// This latest SHA1 fingerprint was updated 13.07.2020
//const char echo_org_ssl_fingerprint[] PROGMEM   = "F0 DC 2E 40 A6 6D 29 B5 73 8F E1 E8 A9 EA 2A 9B 50 68 80 E3";

// This latest SHA1 fingerprint was updated 15.04.2021
const char echo_org_ssl_fingerprint[] PROGMEM   = "34 A2 66 08 A1 4D 1E 83 1A 0E 49 3C 4A 84 45 9E 4A 0D 08 FE";

...

```

I'll update ESP32 full CA Cert later.

khoih-prog, 2021-04-15:
The new CA Cert is as follows, and I check it's OK now. Just replace in defines.h

```
// This certificate was updated 15.04.2021, issues on Mar 15th 2021, expired on June 13th 2021
const char echo_org_ssl_ca_cert[] PROGMEM = \
"-----BEGIN CERTIFICATE-----\n" \
"MIIEZTCCA02gAwIBAgIQQAF1BIMUpMghjISpDBbN3zANBgkqhkiG9w0BAQsFADA/\n" \
"MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT\n" \
"DkRTVCBSb290IENBIFgzMB4XDTIwMTAwNzE5MjE0MFoXDTIxMDkyOTE5MjE0MFow\n" \
"MjELMAkGA1UEBhMCVVMxFjAUBgNVBAoTDUxldCdzIEVuY3J5cHQxCzAJBgNVBAMT\n" \
"AlIzMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAuwIVKMz2oJTTDxLs\n" \
"jVWSw/iC8ZmmekKIp10mqrUrucVMsa+Oa/l1yKPXD0eUFFU1V4yeqKI5GfWCPEKp\n" \
"Tm71O8Mu243AsFzzWTjn7c9p8FoLG77AlCQlh/o3cbMT5xys4Zvv2+Q7RVJFlqnB\n" \
"U840yFLuta7tj95gcOKlVKu2bQ6XpUA0ayvTvGbrZjR8+muLj1cpmfgwF126cm/7\n" \
"gcWt0oZYPRfH5wm78Sv3htzB2nFd1EbjzK0lwYi8YGd1ZrPxGPeiXOZT/zqItkel\n" \
"/xMY6pgJdz+dU/nPAeX1pnAXFK9jpP+Zs5Od3FOnBv5IhR2haa4ldbsTzFID9e1R\n" \
"oYvbFQIDAQABo4IBaDCCAWQwEgYDVR0TAQH/BAgwBgEB/wIBADAOBgNVHQ8BAf8E\n" \
"BAMCAYYwSwYIKwYBBQUHAQEEPzA9MDsGCCsGAQUFBzAChi9odHRwOi8vYXBwcy5p\n" \
"ZGVudHJ1c3QuY29tL3Jvb3RzL2RzdHJvb3RjYXgzLnA3YzAfBgNVHSMEGDAWgBTE\n" \
"p7Gkeyxx+tvhS5B1/8QVYIWJEDBUBgNVHSAETTBLMAgGBmeBDAECATA/BgsrBgEE\n" \
"AYLfEwEBATAwMC4GCCsGAQUFBwIBFiJodHRwOi8vY3BzLnJvb3QteDEubGV0c2Vu\n" \
"Y3J5cHQub3JnMDwGA1UdHwQ1MDMwMaAvoC2GK2h0dHA6Ly9jcmwuaWRlbnRydXN0\n" \
"LmNvbS9EU1RST09UQ0FYM0NSTC5jcmwwHQYDVR0OBBYEFBQusxe3WFbLrlAJQOYf\n" \
"r52LFMLGMB0GA1UdJQQWMBQGCCsGAQUFBwMBBggrBgEFBQcDAjANBgkqhkiG9w0B\n" \
"AQsFAAOCAQEA2UzgyfWEiDcx27sT4rP8i2tiEmxYt0l+PAK3qB8oYevO4C5z70kH\n" \
"ejWEHx2taPDY/laBL21/WKZuNTYQHHPD5b1tXgHXbnL7KqC401dk5VvCadTQsvd8\n" \
"S8MXjohyc9z9/G2948kLjmE6Flh9dDYrVYA9x2O+hEPGOaEOa1eePynBgPayvUfL\n" \
"qjBstzLhWVQLGAkXXmNs+5ZnPBxzDJOLxhF2JIbeQAcH5H0tZrUlo5ZYyOqA7s9p\n" \
"O5b85o3AM/OJ+CktFBQtfvBhcJVd9wvlwPsk+uyOy2HI7mNxKKgsBTt375teA2Tw\n" \
"UdHkhVNcsAKX1H7GNNLOEADksd86wuoXvg==\n" \
"-----END CERTIFICATE-----\n";
```

This is the terminal output showing everything is OK

```
Starting WebSockets Secured-ESP32-Client on ESP32_DEV
WebSockets2_Generic v1.1.0
...
Connected to Wifi, Connecting to WebSockets Server @wss://echo.websocket.org/
[WS] WebsocketsClient::doestStartsWith: str = wss://echo.websocket.org/
[WS] WebsocketsClient::doestStartsWith: prefix = wss://
[WS] WebsocketsClient::connect: wss upgradeToSecuredConnection
[WS] WebsocketsClient::upgradeToSecuredConnection: SSL exit
[WS] WebsocketsClient::connect: step 1
[WS] WebsocketsClient::generateHandshake: base64Authorization = 
[WS] WebsocketsClient::generateHandshake: handshake = GET / HTTP/1.1
Host: echo.websocket.org
Sec-WebSocket-Key: MDEyMzQ1Njc4OWFiY2RlZg==
Upgrade: websocket
Connection: Upgrade
Sec-WebSocket-Version: 13
User-Agent: TinyWebsockets Client
Authorization: Basic 
Origin: https://github.com/khoih-prog/Websockets2_Generic

[WS] WebsocketsClient::connect: base64Authorization = 
[WS] WebsocketsClient::connect: step 2
[WS] WebsocketsClient::connect: step 3
[WS] WebsocketsClient::connect: step 4
[WS] WebsocketsClient::doestStartsWith: str = HTTP/1.1 101 Web Socket Protocol Handshake

[WS] WebsocketsClient::doestStartsWith: prefix = HTTP/1.1 101
[WS] WebsocketsClient::connect: step 5
[WS] WebsocketsClient::connect: step 6
[WS] WebsocketsClient::generateHandshake: key = Access-Control-Allow-Credentials
[WS] WebsocketsClient::generateHandshake: value = true
[WS] WebsocketsClient::generateHandshake: key = Access-Control-Allow-Headers
[WS] WebsocketsClient::generateHandshake: value = content-type
[WS] WebsocketsClient::generateHandshake: key = Access-Control-Allow-Headers
[WS] WebsocketsClient::generateHandshake: value = authorization
[WS] WebsocketsClient::generateHandshake: key = Access-Control-Allow-Headers
[WS] WebsocketsClient::generateHandshake: value = x-websocket-extensions
[WS] WebsocketsClient::generateHandshake: key = Access-Control-Allow-Headers
[WS] WebsocketsClient::generateHandshake: value = x-websocket-version
[WS] WebsocketsClient::generateHandshake: key = Access-Control-Allow-Headers
[WS] WebsocketsClient::generateHandshake: value = x-websocket-protocol
[WS] WebsocketsClient::generateHandshake: key = Access-Control-Allow-Origin
[WS] WebsocketsClient::generateHandshake: value = https://github.com/khoih-prog/Websockets2_Generic
[WS] WebsocketsClient::generateHandshake: key = Connection
[WS] WebsocketsClient::generateHandshake: value = Upgrade
[WS] WebsocketsClient::generateHandshake: key = Date
[WS] WebsocketsClient::generateHandshake: value = Thu, 15 Apr 2021 19:20:27 GMT
[WS] WebsocketsClient::generateHandshake: key = Sec-WebSocket-Accept
[WS] WebsocketsClient::generateHandshake: value = BACScCJPNqyz+UBoqMH89VmURoA=
[WS] WebsocketsClient::generateHandshake: key = Server
[WS] WebsocketsClient::generateHandshake: value = Kaazing Gateway
[WS] WebsocketsClient::generateHandshake: key = Upgrade
[WS] WebsocketsClient::generateHandshake: value = websocket
[WS] WebsocketsClient::connect: step 7
Connnection Opened
Connected!
Got Message: Hello to Server from ESP32_DEV
Got a Pong!
Got a Ping!
Got a Ping!
Got a Ping!
Got a Ping!
Got a Ping!
Got a Ping!
Got a Ping!
Got a Ping!
Got a Ping!
Got a Ping!
```

I'll update a new release soon to update all exaamples.

khoih-prog, 2021-04-15:
This is debug output for ESP8266 using new **fingerprint**

```
Starting Secured-ESP8266-Client on ESP8266_NODEMCU
WebSockets2_Generic v1.1.0
...Connected to Wifi, Connecting to WebSockets Server @wss://echo.websocket.org/
[WS] WebsocketsClient::doestStartsWith: str = wss://echo.websocket.org/
[WS] WebsocketsClient::doestStartsWith: prefix = wss://
[WS] WebsocketsClient::connect: wss upgradeToSecuredConnection
[WS] WebsocketsClient::upgradeToSecuredConnection: SSL exit
[WS] WebsocketsClient::connect: step 1
[WS] WebsocketsClient::generateHandshake: base64Authorization = 
[WS] WebsocketsClient::generateHandshake: handshake = GET / HTTP/1.1
Host: echo.websocket.org
Sec-WebSocket-Key: MDEyMzQ1Njc4OWFiY2RlZg==
Upgrade: websocket
Connection: Upgrade
Sec-WebSocket-Version: 13
User-Agent: TinyWebsockets Client
Authorization: Basic 
Origin: https://github.com/khoih-prog/Websockets2_Generic

[WS] WebsocketsClient::connect: base64Authorization = 
[WS] WebsocketsClient::connect: step 2
[WS] WebsocketsClient::connect: step 3
[WS] WebsocketsClient::connect: step 4
[WS] WebsocketsClient::doestStartsWith: str = HTTP/1.1 101 Web Socket Protocol Handshake

[WS] WebsocketsClient::doestStartsWith: prefix = HTTP/1.1 101
[WS] WebsocketsClient::connect: step 5
[WS] WebsocketsClient::connect: step 6
[WS] WebsocketsClient::generateHandshake: key = Access-Control-Allow-Credentials
[WS] WebsocketsClient::generateHandshake: value = true
[WS] WebsocketsClient::generateHandshake: key = Access-Control-Allow-Headers
[WS] WebsocketsClient::generateHandshake: value = content-type
[WS] WebsocketsClient::generateHandshake: key = Access-Control-Allow-Headers
[WS] WebsocketsClient::generateHandshake: value = authorization
[WS] WebsocketsClient::generateHandshake: key = Access-Control-Allow-Headers
[WS] WebsocketsClient::generateHandshake: value = x-websocket-extensions
[WS] WebsocketsClient::generateHandshake: key = Access-Control-Allow-Headers
[WS] WebsocketsClient::generateHandshake: value = x-websocket-version
[WS] WebsocketsClient::generateHandshake: key = Access-Control-Allow-Headers
[WS] WebsocketsClient::generateHandshake: value = x-websocket-protocol
[WS] WebsocketsClient::generateHandshake: key = Access-Control-Allow-Origin
[WS] WebsocketsClient::generateHandshake: value = https://github.com/khoih-prog/Websockets2_Generic
[WS] WebsocketsClient::generateHandshake: key = Connection
[WS] WebsocketsClient::generateHandshake: value = Upgrade
[WS] WebsocketsClient::generateHandshake: key = Date
[WS] WebsocketsClient::generateHandshake: value = Thu, 15 Apr 2021 19:28:05 GMT
[WS] WebsocketsClient::generateHandshake: key = Sec-WebSocket-Accept
[WS] WebsocketsClient::generateHandshake: value = BACScCJPNqyz+UBoqMH89VmURoA=
[WS] WebsocketsClient::generateHandshake: key = Server
[WS] WebsocketsClient::generateHandshake: value = Kaazing Gateway
[WS] WebsocketsClient::generateHandshake: key = Upgrade
[WS] WebsocketsClient::generateHandshake: value = websocket
[WS] WebsocketsClient::connect: step 7
Connnection Opened
Called Connect to server.
Connected!
Got Message: Hello to Server from ESP8266
Got a Pong!

```

khoih-prog, 2021-04-15:
I guess you have the same Certificate issue with [**webSocket.beginSSL not works on ESP32 but it works on ESP8266** #654](https://github.com/Links2004/arduinoWebSockets/issues/654#issuecomment-820459457)

closed 2021-04-15 by khoih-prog

holmser, 2021-05-12:
How do you get that cert?  Some commands would be helpful.

khoih-prog, 2021-05-12:
I'll write detailed instructions and attach to the libraries, so that easier for you all to find and use.

Basically, you can use the following command (in **Linux, Ubuntu**, etc), and similar in **Windows**

For example, to get the CA Certs for websocket.org (SSL using port 443)

```
$ openssl s_client -showcerts -connect websocket.org:443
```

the output is as follows:

```
CONNECTED(00000003)
depth=2 O = Digital Signature Trust Co., CN = DST Root CA X3
verify return:1
depth=1 C = US, O = Let's Encrypt, CN = R3
verify return:1
depth=0 CN = websocket.org
verify return:1
---
Certificate chain
 0 s:CN = websocket.org
   i:C = US, O = Let's Encrypt, CN = R3
-----BEGIN CERTIFICATE-----
MIIFLzCCBBegAwIBAgISBBcuJaKHZ67E7zWznbuEk/x2MA0GCSqGSIb3DQEBCwUA
MDIxCzAJBgNVBAYTAlVTMRYwFAYDVQQKEw1MZXQncyBFbmNyeXB0MQswCQYDVQQD
EwJSMzAeFw0yMTAzMTUyMTA1NDNaFw0yMTA2MTMyMTA1NDNaMBgxFjAUBgNVBAMT
DXdlYnNvY2tldC5vcmcwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQCX
svTJlKOrIg0Sfoe7JtjW2mMdLYO6CCg3tKHMqIFBH9istUH/jr6XwMPJlXUhPeQt
6gjvxdzNm5B9AC2QVp5X1A7eSKzb9p/pdcF/BFDNY0tDI1TmmZWj4PXhcFV/5Y9b
hXCPq8XONWuFlBJMlew69vqwmgYKrqmUhtTrinCkMQFLJkAU5kZxJt9J20ZrRBrs
J75wtQzQ/q9f9mGt5LopnN4bSZJV0ejLNyKNrKajFZbBT/8JgwIt51TBluT0mJEk
Mgq6+fBOExzcNWxTv13BuIn50ibmkzSkuM92tThWemvhAXIZcSNW2+aPp5iwAAP4
PTa5+miaRUvdFFRDlnAzAgMBAAGjggJXMIICUzAOBgNVHQ8BAf8EBAMCBaAwHQYD
VR0lBBYwFAYIKwYBBQUHAwEGCCsGAQUFBwMCMAwGA1UdEwEB/wQCMAAwHQYDVR0O
BBYEFMvtMY9hWJeWqwcZUTHp3mVylbjbMB8GA1UdIwQYMBaAFBQusxe3WFbLrlAJ
QOYfr52LFMLGMFUGCCsGAQUFBwEBBEkwRzAhBggrBgEFBQcwAYYVaHR0cDovL3Iz
Lm8ubGVuY3Iub3JnMCIGCCsGAQUFBzAChhZodHRwOi8vcjMuaS5sZW5jci5vcmcv
MCkGA1UdEQQiMCCCDyoud2Vic29ja2V0Lm9yZ4INd2Vic29ja2V0Lm9yZzBMBgNV
HSAERTBDMAgGBmeBDAECATA3BgsrBgEEAYLfEwEBATAoMCYGCCsGAQUFBwIBFhpo
dHRwOi8vY3BzLmxldHNlbmNyeXB0Lm9yZzCCAQIGCisGAQQB1nkCBAIEgfMEgfAA
7gB1AESUZS6w7s6vxEAH2Kj+KMDa5oK+2MsxtT/TM5a1toGoAAABeDfsauMAAAQD
AEYwRAIgUParUiaMDb/7aXk9WWcm+DNu6GNp6LJGlxkIuoFL+78CIE34ajis+a58
U29HG6Wyt04/+Yuli4SlEEmBzGfwGPTKAHUAfT7y+I//iFVoJMLAyp5SiXkrxQ54
CX8uapdomX4i8NcAAAF4N+xq+gAABAMARjBEAiBsQqJjss/ggVnUaXVXhxuDEBTn
ZzlUnrliuvoUi4arIwIgOcylAEWEGmuGL+gxiV4dfLZwjWGdYnKSbecvB8LkbbQw
DQYJKoZIhvcNAQELBQADggEBAKmEDjhxQTj+sBgPvf4kmamaQElfDJQiTanJwIKk
k9N/IHCfzsGMdfayEKPVWluF0yWkR7dEkVdx3vqwB4XnlPWjDZCZKx5hGGBZnPb/
Bw+7iwyFrEgB+n4yr7GmZRiM4WunguLWxBKB8oTDn5Nf7UnWLo9L2/4kJSeuxqns
ACF0GJharBm5imHGviGWXCuYlVfepRq1b0mBbliS/vWHVvgGA8FwfgE/gW5j6hJU
xC4ghthAgqiauEJoHz+IrlsBiUHBos/Lib7r14FdVTTNAli1e7uDVP6yxcRUisTd
PxHQdHTd9Rq+6lBLGNmFdfB/A17p5py+MyWXtju0Oewnss8=
-----END CERTIFICATE-----
 1 s:C = US, O = Let's Encrypt, CN = R3
   i:O = Digital Signature Trust Co., CN = DST Root CA X3
-----BEGIN CERTIFICATE-----
MIIEZTCCA02gAwIBAgIQQAF1BIMUpMghjISpDBbN3zANBgkqhkiG9w0BAQsFADA/
MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT
DkRTVCBSb290IENBIFgzMB4XDTIwMTAwNzE5MjE0MFoXDTIxMDkyOTE5MjE0MFow
MjELMAkGA1UEBhMCVVMxFjAUBgNVBAoTDUxldCdzIEVuY3J5cHQxCzAJBgNVBAMT
AlIzMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAuwIVKMz2oJTTDxLs
jVWSw/iC8ZmmekKIp10mqrUrucVMsa+Oa/l1yKPXD0eUFFU1V4yeqKI5GfWCPEKp
Tm71O8Mu243AsFzzWTjn7c9p8FoLG77AlCQlh/o3cbMT5xys4Zvv2+Q7RVJFlqnB
U840yFLuta7tj95gcOKlVKu2bQ6XpUA0ayvTvGbrZjR8+muLj1cpmfgwF126cm/7
gcWt0oZYPRfH5wm78Sv3htzB2nFd1EbjzK0lwYi8YGd1ZrPxGPeiXOZT/zqItkel
/xMY6pgJdz+dU/nPAeX1pnAXFK9jpP+Zs5Od3FOnBv5IhR2haa4ldbsTzFID9e1R
oYvbFQIDAQABo4IBaDCCAWQwEgYDVR0TAQH/BAgwBgEB/wIBADAOBgNVHQ8BAf8E
BAMCAYYwSwYIKwYBBQUHAQEEPzA9MDsGCCsGAQUFBzAChi9odHRwOi8vYXBwcy5p
ZGVudHJ1c3QuY29tL3Jvb3RzL2RzdHJvb3RjYXgzLnA3YzAfBgNVHSMEGDAWgBTE
p7Gkeyxx+tvhS5B1/8QVYIWJEDBUBgNVHSAETTBLMAgGBmeBDAECATA/BgsrBgEE
AYLfEwEBATAwMC4GCCsGAQUFBwIBFiJodHRwOi8vY3BzLnJvb3QteDEubGV0c2Vu
Y3J5cHQub3JnMDwGA1UdHwQ1MDMwMaAvoC2GK2h0dHA6Ly9jcmwuaWRlbnRydXN0
LmNvbS9EU1RST09UQ0FYM0NSTC5jcmwwHQYDVR0OBBYEFBQusxe3WFbLrlAJQOYf
r52LFMLGMB0GA1UdJQQWMBQGCCsGAQUFBwMBBggrBgEFBQcDAjANBgkqhkiG9w0B
AQsFAAOCAQEA2UzgyfWEiDcx27sT4rP8i2tiEmxYt0l+PAK3qB8oYevO4C5z70kH
ejWEHx2taPDY/laBL21/WKZuNTYQHHPD5b1tXgHXbnL7KqC401dk5VvCadTQsvd8
S8MXjohyc9z9/G2948kLjmE6Flh9dDYrVYA9x2O+hEPGOaEOa1eePynBgPayvUfL
qjBstzLhWVQLGAkXXmNs+5ZnPBxzDJOLxhF2JIbeQAcH5H0tZrUlo5ZYyOqA7s9p
O5b85o3AM/OJ+CktFBQtfvBhcJVd9wvlwPsk+uyOy2HI7mNxKKgsBTt375teA2Tw
UdHkhVNcsAKX1H7GNNLOEADksd86wuoXvg==
-----END CERTIFICATE-----
---
Server certificate
subject=CN = websocket.org

issuer=C = US, O = Let's Encrypt, CN = R3

---
No client certificate CA names sent
---
SSL handshake has read 2638 bytes and written 647 bytes
Verification: OK
---
New, SSLv3, Cipher is AES128-SHA
Server public key is 2048 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : AES128-SHA
    Session-ID: 609C4B74C7154E1DECA4694064589FFC63538BB77B9CB353B6D6928EFE506745
    Session-ID-ctx: 
    Master-Key: 5AF98064A9D58C2B5F9C37B4CD1786C8D0949DAC6F9E01B5E1F2EAA2FC27D3B73C50F8F60413D201EC4A2C2E095F6C43
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    Start Time: 1620856092
    Timeout   : 7200 (sec)
    Verify return code: 0 (ok)
    Extended master secret: no
---
q
closed
```
---

We'll use the second Cert.
Depending on which format to use, you can either manually modify or write a script to convert. 
For example, from

```
...
MIIEZTCCA02gAwIBAgIQQAF1BIMUpMghjISpDBbN3zANBgkqhkiG9w0BAQsFADA/
...
```

to

```
...
"MIIEZTCCA02gAwIBAgIQQAF1BIMUpMghjISpDBbN3zANBgkqhkiG9w0BAQsFADA/\n" \
...
```

Happy exploring,