khoih-prog/WebSockets2_Generic #23: Websocket example with esp32 and Lan8720

andrettaadriano1, 2021-07-05:
can you insert examples of shetch with websocket and secure wesocket (wss) working in esp32 with ethernet via lan8720?

khoih-prog, 2021-07-05:
Sorry, ESP32 + LAN8720 (such as **WT32-ETH01**) is not currently supported, and I currently have no plan to provide support to the still rare combination.

I'll, anyway, definitely spend some time to add support to many underlying libraries, on which this library depends, before moving on to this library. Certainly, the job will have higher priority if more requests are received.

I'm closing the issue now, and will provide update whenever the support is ready.

Good Luck,

closed 2021-07-05 by khoih-prog

andrettaadriano1, 2021-07-06:
what combination of microcontroller / ethernet Ic do you recommend to use your library via ethernet?

khoih-prog, 2021-07-06:
I'm still thinking the ESP32 + LAN8720 (such as WT32-ETH01) is the best choice, for **cost, power, reliability and plenty of support** (in the near future when I finish some libraries to support it).

I'm currently working and nearly finish the first underlying library, named **WebServer_WT32_ETH01**, to provide similar features as [EthernetWebServer library](https://github.com/khoih-prog/EthernetWebServer). It'll be released by tomorrow. Then move on to other libraries, such as WebSockets, etc.

So, if you can wait, those libraries will be finished within 1-2 weeks.

Otherwise, you can select some boards, such as the new and cheap RP2040 + W5500, or ESP32 + W5500, ESP8266 + W5500, STM32F4 + LAN8720, etc.

khoih-prog, 2021-07-06:
You can try the first library of the chain :  [**WebServer_WT32_ETH01**](https://github.com/khoih-prog/WebServer_WT32_ETH01) which provides the same features of [**EthernetWebServer**](https://github.com/khoih-prog/EthernetWebServer).

khoih-prog, 2021-07-08:
The new [**WebSockets2_Generic** Releases v1.5.0](https://github.com/khoih-prog/WebSockets2_Generic/releases/tag/v1.5.0) has just been published to provide support to WT32_ETH01 (ESP32 + LAN8720) boards.

Try the examples in [**WT32_ETH01**](https://github.com/khoih-prog/WebSockets2_Generic/tree/master/examples/WT32_ETH01) written specially for ESP32 + LAN8720.

---

### Major Release v1.5.0

1. Add support to **WT32_ETH01 (ESP32 + LAN8720)** boards
2. Add examples with new features