khoih-prog/WebSockets2_Generic #25: Numerous errors logged when compiling, not sure what the actual issue is.

jackmerrill, 2021-07-23:
Hi there! Whenever I build my code, I get multiple errors:

```
Arduino: 1.8.15 (Mac OS X), Board: "Arduino NANO 33 IoT"

In file included from /Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic.h:49:0,
                 from /Users/jackmerrill/Documents/Arduino/Friendship_Lamp/Friendship_Lamp.ino:9:
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/Tiny_Websockets_Generic/message.hpp:72:4: warning: #warning WEBSOCKETS_USE_ESP_WIFI in message.hpp [-Wcpp]
   #warning WEBSOCKETS_USE_ESP_WIFI in message.hpp
    ^~~~~~~
In file included from /Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic.h:50:0,
                 from /Users/jackmerrill/Documents/Arduino/Friendship_Lamp/Friendship_Lamp.ino:9:
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/Tiny_Websockets_Generic/client.hpp:78:4: warning: #warning WEBSOCKETS_USE_ESP_WIFI in client.hpp [-Wcpp]
   #warning WEBSOCKETS_USE_ESP_WIFI in client.hpp
    ^~~~~~~
In file included from /Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic.h:51:0,
                 from /Users/jackmerrill/Documents/Arduino/Friendship_Lamp/Friendship_Lamp.ino:9:
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/Tiny_Websockets_Generic/server.hpp:73:4: warning: #warning WEBSOCKETS_USE_ESP_WIFI in server.hpp [-Wcpp]
   #warning WEBSOCKETS_USE_ESP_WIFI in server.hpp
    ^~~~~~~
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/Tiny_Websockets_Generic/server.hpp:83:51: error: expected ')' before '*' token
       WebsocketsServer(network2_generic::TcpServer* server = new WSDefaultTcpServer);
                                                   ^
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/Tiny_Websockets_Generic/server.hpp:99:25: error: 'TcpServer' in namespace 'websockets2_generic::network2_generic' does not name a type
       network2_generic::TcpServer* _server;
                         ^~~~~~~~~
In file included from /Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic.h:54:0,
                 from /Users/jackmerrill/Documents/Arduino/Friendship_Lamp/Friendship_Lamp.ino:9:
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic_Client.hpp: In constructor 'websockets2_generic::WebsocketsClient::WebsocketsClient()':
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic_Client.hpp:54:76: error: 'WSDefaultTcpClient' was not declared in this scope
   WebsocketsClient::WebsocketsClient() : WebsocketsClient(std::make_shared<WSDefaultTcpClient>())
                                                                            ^~~~~~~~~~~~~~~~~~
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic_Client.hpp:54:96: error: no matching function for call to 'make_shared<<expression error> >()'
   WebsocketsClient::WebsocketsClient() : WebsocketsClient(std::make_shared<WSDefaultTcpClient>())
                                                                                                ^
In file included from /Users/jackmerrill/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/memory:81:0,
                 from /Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/Tiny_Websockets_Generic/internals/websockets_endpoint.hpp:42,
                 from /Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/Tiny_Websockets_Generic/client.hpp:43,
                 from /Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic.h:50,
                 from /Users/jackmerrill/Documents/Arduino/Friendship_Lamp/Friendship_Lamp.ino:9:
/Users/jackmerrill/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/bits/shared_ptr.h:703:5: note: candidate: template<class _Tp, class ... _Args> std::shared_ptr<_Tp> std::make_shared(_Args&& ...)
     make_shared(_Args&&... __args)
     ^~~~~~~~~~~
/Users/jackmerrill/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/bits/shared_ptr.h:703:5: note:   template argument deduction/substitution failed:
In file included from /Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic.h:54:0,
                 from /Users/jackmerrill/Documents/Arduino/Friendship_Lamp/Friendship_Lamp.ino:9:
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic_Client.hpp:54:96: error: template argument 1 is invalid
   WebsocketsClient::WebsocketsClient() : WebsocketsClient(std::make_shared<WSDefaultTcpClient>())
                                                                                                ^
In file included from /Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic.h:54:0,
                 from /Users/jackmerrill/Documents/Arduino/Friendship_Lamp/Friendship_Lamp.ino:9:
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic_Client.hpp: In member function 'void websockets2_generic::WebsocketsClient::upgradeToSecuredConnection()':
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic_Client.hpp:398:23: error: 'WSDefaultSecuredTcpClient' does not name a type
     auto client = new WSDefaultSecuredTcpClient;
                       ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic_Client.hpp:464:37: error: 'WSDefaultSecuredTcpClient' was not declared in this scope
     this->_client = std::shared_ptr<WSDefaultSecuredTcpClient>(client);
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic_Client.hpp:464:62: error: template argument 1 is invalid
     this->_client = std::shared_ptr<WSDefaultSecuredTcpClient>(client);
                                                              ^
In file included from /Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic.h:55:0,
                 from /Users/jackmerrill/Documents/Arduino/Friendship_Lamp/Friendship_Lamp.ino:9:
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic_Server.hpp: At global scope:
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic_Server.hpp:52:37: error: expected constructor, destructor, or type conversion before '(' token
   WebsocketsServer::WebsocketsServer(network2_generic::TcpServer* server) : _server(server) {}
                                     ^
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic_Server.hpp: In member function 'bool websockets2_generic::WebsocketsServer::available()':
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic_Server.hpp:56:18: error: 'class websockets2_generic::WebsocketsServer' has no member named '_server'
     return this->_server->available();
[... 9 lines cut ...]
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic_Server.hpp: In member function 'websockets2_generic::WebsocketsClient websockets2_generic::WebsocketsServer::accept()':
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic_Server.hpp:123:60: error: '_server' was not declared in this scope
     std::shared_ptr<network2_generic::TcpClient> tcpClient(_server->accept());
                                                            ^~~~~~~
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic_Server.hpp: In destructor 'virtual websockets2_generic::WebsocketsServer::~WebsocketsServer()':
/Users/jackmerrill/Documents/Arduino/libraries/WebSockets2_Generic-master/src/WebSockets2_Generic_Server.hpp:193:11: error: 'class websockets2_generic::WebsocketsServer' has no member named '_server'
     this->_server->close();
           ^~~~~~~
exit status 1
[... cut ...]
I'm a bit of an Arduino noob so I'm not exactly sure how to fix this (yes my project is a bit ambitious, but I can pull it off).

Here's my project's code repository: https://github.com/jackmerrill/FriendshipLamp

I copied some code from the Generic WifiNINA SAMD project, https://github.com/khoih-prog/WebSockets2_Generic/tree/master/examples/Generic/WiFiNINA/SAMD/SAMD-Client

the `EasyWifi` is [another repository](https://github.com/javos65/EasyWifi-for-MKR1010) that I slightly modified in an attempt to get it working with WiFiNINA_Generic.

Any and all help is appreciated, thank you!

khoih-prog, 2021-07-23:
Just a small change at the start of [Friendship_Lamp.ino](https://github.com/jackmerrill/FriendshipLamp/blob/main/Friendship_Lamp.ino)

```
#include "defines.h"

#include <Adafruit_NeoPixel.h>
#include "EasyWiFi.h"
#include <SPI.h>
#include <WebSockets2_Generic.h>

using namespace websockets2_generic;
```

[image]

Good Luck with your project !
I'd appreciate it if you can post here after you've finished your good project so that other users and I can use ;-)

closed 2021-07-23 by khoih-prog

jackmerrill, 2021-07-23:
AHHH Thank you!

khoih-prog, 2021-07-23:
Wow, you're just **a 16 year old Full-Stack developer from Chicago.**

I'm amazed at the quality of your code. You're such a **genius** at your age. Keep up with the great work so far and help the digital world.

Best Regards,

jackmerrill, 2021-07-23:
Haha thank you! I'm pretty experienced with other languages so this isn't **too** hard.