khoih-prog/WebSockets2_Generic #32: Compile error with Arduino Nano RP2040

Sirpercal, 2021-10-03:
Arduino:1.8.15 Hourly Build 2021/05/19 12:35 (Mac OS X 10.14.6), Board : "**_Arduino Nano RP2040 Connect_**"
mbed_nano/2.4.1

I've try to compile the simple WebSocket Client example with the changes:

````
#if ( defined(ARDUINO_ARDUINO_NANO33BLE) || defined(ARDUINO_ARCH_SAMD) || defined(ARDUINO_ARCH_MBED_RP2040) || defined(ARDUINO_ARCH_RP2040) )
#define WEBSOCKETS_USE_WIFININA           true
#else
#error This code is intended to run only on the RP2040 boards ! Please check your Tools->Board setting.
#endif

#if defined(WEBSOCKETS_WIFININA_USE_NRF52)

#if defined(ARDUINO_ARDUINO_NANO33BLE)
#define BOARD_TYPE      "ARDUINO_ARDUINO_NANO33BLE"
#elif defined(ARDUINO_ARCH_SAMD)
#define BOARD_TYPE      "ARDUINO_ARCH_SAMD"
#elif defined(ARDUINO_ARCH_MBED_RP2040)
#define BOARD_TYPE      "ARDUINO_ARCH_MBED_RP2040"
#elif defined(ARDUINO_ARCH_RP2040)
#define BOARD_TYPE      "ARDUINO_ARCH_RP2040"
#else
#define BOARD_TYPE      "RP2040 Unknown"
#endif

#endif
````
And get this compile error:
```
Arduino:1.8.15 Hourly Build 2021/05/19 12:35 (Mac OS X), Carte : "Arduino Nano RP2040 Connect"

In file included from /Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic.h:53:0,
                 from /Users/bobby/Documents/Arduino/RP2040-Client-Websocket2_Generic/RP2040-Client-Websocket2_Generic.ino:34:
/Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/Tiny_Websockets_Generic/server.hpp:96:51: error: expected ')' before '*' token
       WebsocketsServer(network2_generic::TcpServer* server = new WSDefaultTcpServer);
                                                   ^
/Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/Tiny_Websockets_Generic/server.hpp:112:25: error: 'TcpServer' in namespace 'websockets2_generic::network2_generic' does not name a type
       network2_generic::TcpServer* _server;
                         ^~~~~~~~~
In file included from /Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic.h:56:0,
                 from /Users/bobby/Documents/Arduino/RP2040-Client-Websocket2_Generic/RP2040-Client-Websocket2_Generic.ino:34:
/Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic_Client.hpp: In constructor 'websockets2_generic::WebsocketsClient::WebsocketsClient()':
/Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic_Client.hpp:56:76: error: 'WSDefaultTcpClient' was not declared in this scope
   WebsocketsClient::WebsocketsClient() : WebsocketsClient(std::make_shared<WSDefaultTcpClient>())
                                                                            ^~~~~~~~~~~~~~~~~~
/Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic_Client.hpp:56:96: error: no matching function for call to 'make_shared<<expression error> >()'
   WebsocketsClient::WebsocketsClient() : WebsocketsClient(std::make_shared<WSDefaultTcpClient>())
                                                                                                ^
In file included from /Users/bobby/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/memory:81:0,
                 from /Users/bobby/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.4.1/cores/arduino/mbed/platform/cxxsupport/mstd_memory:34,
                 from /Users/bobby/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.4.1/cores/arduino/mbed/platform/cxxsupport/mstd_functional:31,
                 from /Users/bobby/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.4.1/cores/arduino/mbed/platform/include/platform/Callback.h:27,
                 from /Users/bobby/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.4.1/cores/arduino/mbed/platform/include/platform/FileHandle.h:23,
                 from /Users/bobby/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.4.1/cores/arduino/macros.h:41,
                 from /Users/bobby/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.4.1/variants/NANO_RP2040_CONNECT/pins_arduino.h:2,
                 from /Users/bobby/Library/Arduino15/packages/arduino/hardware/mbed_nano/2.4.1/cores/arduino/Arduino.h:76,
                 from sketch/RP2040-Client-Websocket2_Generic.ino.cpp:1:
/Users/bobby/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/bits/shared_ptr.h:703:5: note: candidate: template<class _Tp, class ... _Args> std::shared_ptr<_Tp> std::make_shared(_Args&& ...)
     make_shared(_Args&&... __args)
     ^~~~~~~~~~~
/Users/bobby/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/bits/shared_ptr.h:703:5: note:   template argument deduction/substitution failed:
In file included from /Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic.h:56:0,
                 from /Users/bobby/Documents/Arduino/RP2040-Client-Websocket2_Generic/RP2040-Client-Websocket2_Generic.ino:34:
/Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic_Client.hpp:56:96: error: template argument 1 is invalid
   WebsocketsClient::WebsocketsClient() : WebsocketsClient(std::make_shared<WSDefaultTcpClient>())
                                                                                                ^
In file included from /Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic.h:56:0,
                 from /Users/bobby/Documents/Arduino/RP2040-Client-Websocket2_Generic/RP2040-Client-Websocket2_Generic.ino:34:
/Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic_Client.hpp: In member function 'void websockets2_generic::WebsocketsClient::upgradeToSecuredConnection()':
[... 13 lines cut ...]
   WebsocketsServer::WebsocketsServer(network2_generic::TcpServer* server) : _server(server) {}
                                     ^
/Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic_Server.hpp: In member function 'bool websockets2_generic::WebsocketsServer::available()':
/Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic_Server.hpp:58:18: error: 'class websockets2_generic::WebsocketsServer' has no member named '_server'
     return this->_server->available();
                  ^~~~~~~
/Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic_Server.hpp: In member function 'void websockets2_generic::WebsocketsServer::listen(uint16_t)':
/Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic_Server.hpp:63:11: error: 'class websockets2_generic::WebsocketsServer' has no member named '_server'
     this->_server->listen(port);
           ^~~~~~~
/Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic_Server.hpp: In member function 'bool websockets2_generic::WebsocketsServer::poll()':
/Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic_Server.hpp:68:18: error: 'class websockets2_generic::WebsocketsServer' has no member named '_server'
     return this->_server->poll();
                  ^~~~~~~
/Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic_Server.hpp: In member function 'websockets2_generic::WebsocketsClient websockets2_generic::WebsocketsServer::accept()':
/Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic_Server.hpp:125:60: error: '_server' was not declared in this scope
     std::shared_ptr<network2_generic::TcpClient> tcpClient(_server->accept());
                                                            ^~~~~~~
/Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic_Server.hpp: In destructor 'virtual websockets2_generic::WebsocketsServer::~WebsocketsServer()':
/Users/bobby/Documents/Arduino/libraries/WebSockets2_Generic/src/WebSockets2_Generic_Server.hpp:195:11: error: 'class websockets2_generic::WebsocketsServer' has no member named '_server'
     this->_server->close();
           ^~~~~~~
exit status 1

```

khoih-prog, 2021-10-03:
Hi Robert,

Thanks for your interest in the library.

Currently, `Nano_RP2040_Connect` board is not supported, as somehow I forgot to add its support and will do it ASAP.
Some modifications in the library, in addition to those in the sketch, are necessary.

Will let you know soon,

Best,

khoih-prog, 2021-10-04:
Hi Robert,

The new [**WebSockets2_Generic** releases v1.8.0](https://github.com/khoih-prog/WebSockets2_Generic/releases/tag/v1.8.0) has just been published.

Check the new [**RP2040** examples](https://github.com/khoih-prog/WebSockets2_Generic/tree/master/examples/Generic/WiFiNINA/RP2040)

Your contribution has been noted in [**Contributions and Thanks**](https://github.com/khoih-prog/WebSockets2_Generic#contributions-and-thanks)

Please test and report any issue

Best Regards,

---

### Major Release v1.8.0

1. Add support to **RP2040-based boards, such as Nano_RP2040_Connect, RASPBERRY_PI_PICO, ADAFRUIT_FEATHER_RP2040**, using WiFiNINA
2. Add examples with new features
3. Update many Packages' Patches

closed 2021-10-04 by khoih-prog