khoih-prog/WebSockets2_Generic #36: CA Certificate not getting verified ?

gb-123-git, 2022-01-27:
### Describe the bug

The WebSocket Client doesn't seem to verify the CA Certificate even when one is provided through "client.setCACert(ssl_ca_cert)" method.

### Steps to Reproduce

1. Enter Correct Certificate to the Sketch (Or just use the git provided example).
2. Change a few letters of the certificate.
3. Try to connect, and the client will connect to the server.

PS: client.setInsecure() method had NOT been used;

### Expected behavior

Client should not connect to the Server.

### Actual behavior

Client connects to the Server.

### Information

Please ensure to specify the following:

* PIO Version : Core : 5.2.4, Home 3.4.0
* ESP32 - Wrover

*PLATFORM: Espressif 32 (3.4.0) > Espressif ESP32-DEVKITC-VE (WROVER) CUSTOM
*PACKAGES:
* - framework-arduinoespressif32 0.0.0+sha.fdec056
* - framework-espidf 3.40001.200521 (4.0.1)

khoih-prog, 2022-01-28:
Hi @gb-123-git 

Thanks for your interest in the library.

As you know this library is just a modified version of [**ArduinoWebsockets**](https://github.com/gilmaimon/ArduinoWebsockets) to provide support to boards other than **ESP32/ESP8266**. I suggest that you post your ESP32-related issue on the [**ArduinoWebsockets**](https://github.com/gilmaimon/ArduinoWebsockets) to have a better and faster response.

As the `wss://echo.websocket.org` has been deprecated, please also specify how you tested using the example (which server, certs, etc.)

Just remember to post with [**MRE**](https://stackoverflow.com/help/minimal-reproducible-example) so that anybody can duplicate the issue, or I'm afraid it'll be ignored.

Good Luck,

gb-123-git, 2022-01-28:
I used the LetsEncrypt Cert (Publicly available) on SocketsBay(https://socketsbay.com/test-websockets). I'm afraid their server is also down now, so for replicating the issue one would need an own WebSocket Server. 

I'm afraid I wont be able to provide an MRE. The simplest way to reproduce for people using the code is to change 1-2 digits of the certificate they are using and try to connect to their server. The library instead of verifying the CA Certificate against the Server, would simply allow the connection to go through.

Thanks for your prompt reply and your good work.

khoih-prog, 2022-01-28:
I think you didn't create the correct local SSL WebSocket Server (WSS) because you can't get certificate for local IP addresses

> Issuance of certificates to reserved IP addresses is not allowed, and all certificates previously issued to reserved IP addresses were revoked as of 1 October 2016.

That's why no SSL check is done by the WSS server.

Anyway, I'm closing this now.

closed 2022-01-28 by khoih-prog