khoih-prog/WebSockets2_Generic #39: QNEthernet and NativeEthernet staticIP not working with WS Server

khoih-prog, 2022-03-14:
The original issue was reported in PJRC forum in [WebSockets_Generic Library](https://forum.pjrc.com/threads/61149-WebSockets_Generic-Library)

Duplicated here for easier to follow-up.

1. DHCP is working OK with QNEthernet

```
Starting Teensy41_Client on TEENSY 4.1 using QNEthernet
WebSockets2_Generic v1.10.1
=========== USE_QN_ETHERNET ===========
Initialize Ethernet using DHCP => Connected! IP address:192.168.2.107
[WS] WebsocketsClient::doestStartsWith: str = ws://192.168.2.30:8080
[WS] WebsocketsClient::doestStartsWith: prefix = ws://
[WS] WebsocketsClient::connect: step 1
[WS] WebsocketsClient::generateHandshake: base64Authorization = 
[WS] WebsocketsClient::generateHandshake: handshake = GET / HTTP/1.1
Host: 192.168.2.30
Sec-WebSocket-Key: MDEyMzQ1Njc4OWFiY2RlZg==
Upgrade: websocket
Connection: Upgrade
Sec-WebSocket-Version: 13
User-Agent: TinyWebsockets Client
Authorization: Basic 
Origin: https://github.com/khoih-prog/Websockets2_Generic

[WS] WebsocketsClient::connect: base64Authorization = 
[WS] WebsocketsClient::connect: step 2
[WS] WebsocketsClient::connect: step 3
[WS] WebsocketsClient::connect: step 4
[WS] WebsocketsClient::doestStartsWith: str = HTTP/1.1 101 Switching Protocols

[WS] WebsocketsClient::doestStartsWith: prefix = HTTP/1.1 101
[WS] WebsocketsClient::connect: step 5
[WS] WebsocketsClient::connect: step 6
[WS] WebsocketsClient::generateHandshake: key = Upgrade
[WS] WebsocketsClient::generateHandshake: value = websocket
[WS] WebsocketsClient::generateHandshake: key = Connection
[WS] WebsocketsClient::generateHandshake: value = Upgrade
[WS] WebsocketsClient::generateHandshake: key = Sec-WebSocket-Accept
[WS] WebsocketsClient::generateHandshake: value = BACScCJPNqyz+UBoqMH89VmURoA=
[WS] WebsocketsClient::generateHandshake: key = Origin
[WS] WebsocketsClient::generateHandshake: value = https://github.com/khoih-prog/Websockets2_Generic
[WS] WebsocketsClient::connect: step 7
Connected to server : ws://192.168.2.30:8080
Got Message: Hello Server
```

2. Static IP is not working OK with QNEthernet

```
Starting Teensy41_Client on TEENSY 4.1 using QNEthernet
WebSockets2_Generic v1.10.1
=========== USE_QN_ETHERNET ===========
Initialize Ethernet using static IP => Connected! IP address:192.168.2.222
[WS] WebsocketsClient::doestStartsWith: str = ws://192.168.2.30:8080
[WS] WebsocketsClient::doestStartsWith: prefix = ws://
[WS] WebsocketsClient::connect: step 1
[WS] WebsocketsClient::connect: not _connectionOpen, host = 192.168.2.30 , port = 8080
Couldn't connect to server : ws://192.168.2.30:8080
```

khoih-prog, 2022-03-14:
Test using RP2040 and [**RP2040-Ethernet-Client**](https://github.com/khoih-prog/WebSockets2_Generic/tree/master/examples/Generic/Ethernet/RP2040/RP2040-Ethernet-Client), core arduino-pico, W5500 Ethernet. **Both** DCHP and StaticIP are OK.

### 1. RP2040-Ethernet-Client on RASPBERRY_PI_PICO, DHCP

```
Starting RP2040-Ethernet-Client on RASPBERRY_PI_PICO
Ethernet using W5x00 and EthernetLarge Library
WebSockets2_Generic v1.10.1
_pinCS = 0
W5100 init, using SS_PIN_DEFAULT = 10, new ss_pin = 10, W5100Class::ss_pin = 17
W5100::init: W5500, SSIZE =8192
WebSockets Client IP address: 192.168.2.108      <======  DHCP
Connecting to WebSockets Server @192.168.2.30
[WS] WebsocketsClient::connect: step 1
[WS] WebsocketsClient::generateHandshake: base64Authorization = 
[WS] WebsocketsClient::generateHandshake: handshake = GET / HTTP/1.1
Host: 192.168.2.30
Sec-WebSocket-Key: MDEyMzQ1Njc4OWFiY2RlZg==
Upgrade: websocket
Connection: Upgrade
Sec-WebSocket-Version: 13
User-Agent: TinyWebsockets Client
Authorization: Basic 
Origin: https://github.com/khoih-prog/Websockets2_Generic

[WS] WebsocketsClient::connect: base64Authorization = 
[WS] WebsocketsClient::connect: step 2
[WS] WebsocketsClient::connect: step 3
[WS] WebsocketsClient::connect: step 4
[WS] WebsocketsClient::doestStartsWith: str = HTTP/1.1 101 Switching Protocols

[WS] WebsocketsClient::doestStartsWith: prefix = HTTP/1.1 101
[WS] WebsocketsClient::connect: step 5
[WS] WebsocketsClient::connect: step 6
[WS] WebsocketsClient::generateHandshake: key = Upgrade
[WS] WebsocketsClient::generateHandshake: value = websocket
[WS] WebsocketsClient::generateHandshake: key = Connection
[WS] WebsocketsClient::generateHandshake: value = Upgrade
[WS] WebsocketsClient::generateHandshake: key = Sec-WebSocket-Accept
[WS] WebsocketsClient::generateHandshake: value = BACScCJPNqyz+UBoqMH89VmURoA=
[WS] WebsocketsClient::generateHandshake: key = Origin
[WS] WebsocketsClient::generateHandshake: value = https://github.com/khoih-prog/Websockets2_Generic
[WS] WebsocketsClient::connect: step 7
Connnection Opened
Connected!
Got Message: Hello to Server from RASPBERRY_PI_PICO
Got a Ping!
```

### 2. RP2040-Ethernet-Client on RASPBERRY_PI_PICO, staticIP

```
Starting RP2040-Ethernet-Client on RASPBERRY_PI_PICO
Ethernet using W5x00 and EthernetLarge Library
WebSockets2_Generic v1.10.1

W5100 init, using SS_PIN_DEFAULT = 10, new ss_pin = 10, W5100Class::ss_pin = 17
W5100::init: W5500, SSIZE =8192
WebSockets Client IP address: 192.168.2.222           <==========  Static IP
Connecting to WebSockets Server @192.168.2.30
[WS] WebsocketsClient::connect: step 1
[WS] WebsocketsClient::generateHandshake: base64Authorization = 
[WS] WebsocketsClient::generateHandshake: handshake = GET / HTTP/1.1
Host: 192.168.2.30
Sec-WebSocket-Key: MDEyMzQ1Njc4OWFiY2RlZg==
Upgrade: websocket
Connection: Upgrade
Sec-WebSocket-Version: 13
User-Agent: TinyWebsockets Client
Authorization: Basic 
Origin: https://github.com/khoih-prog/Websockets2_Generic

[WS] WebsocketsClient::connect: base64Authorization = 
[WS] WebsocketsClient::connect: step 2
[WS] WebsocketsClient::connect: step 3
[WS] WebsocketsClient::connect: step 4
[WS] WebsocketsClient::doestStartsWith: str = HTTP/1.1 101 Switching Protocols

[WS] WebsocketsClient::doestStartsWith: prefix = HTTP/1.1 101
[WS] WebsocketsClient::connect: step 5
[WS] WebsocketsClient::connect: step 6
[WS] WebsocketsClient::generateHandshake: key = Upgrade
[WS] WebsocketsClient::generateHandshake: value = websocket
[WS] WebsocketsClient::generateHandshake: key = Connection
[WS] WebsocketsClient::generateHandshake: value = Upgrade
[WS] WebsocketsClient::generateHandshake: key = Sec-WebSocket-Accept
[WS] WebsocketsClient::generateHandshake: value = BACScCJPNqyz+UBoqMH89VmURoA=
[WS] WebsocketsClient::generateHandshake: key = Origin
[WS] WebsocketsClient::generateHandshake: value = https://github.com/khoih-prog/Websockets2_Generic
[WS] WebsocketsClient::connect: step 7
Connnection Opened
Connected!
Got Message: Hello to Server from RASPBERRY_PI_PICO
```

khoih-prog, 2022-03-14:
Hi @ssilverman 

I narrow down the bug is in **QNEthernet and NativeEthernet**. This happens when we're using static and server as numeric IP. If server is using hostname => OK

Steps to duplicate

1. Using example [**WebClient**](https://github.com/khoih-prog/QNEthernet/tree/master/examples/WebClient) from my forked copy of your QNEthernet
2. Change [WebClient.ino#L26](https://github.com/khoih-prog/QNEthernet/blob/master/examples/WebClient/WebClient.ino#L26) to use staticIP

```
#define USING_DHCP    false   //true
```

3. If we keep [WebClient.ino#L23](https://github.com/khoih-prog/QNEthernet/blob/master/examples/WebClient/WebClient.ino#L23) as is to use hostname 

- code
```
char server[] = "arduino.cc";
//IPAddress server(100,24,172,113);
```
then OK

- terminal

```
Starting WebClient using QNEthernet Library
Initialize Ethernet using static IP => IP Address = 192.168.2.222
Connecting to arduino.cc...
Connected to 100.24.172.113
HTTP/1.1 200 OK
Server: nginx/1.4.2
Date: Mon, 14 Mar 2022 03:31:22 GMT
Content-Type: text/plain
Content-Length: 2317
Last-Modified: Thu, 24 Feb 2022 11:33:32 GMT
Connection: close
Vary: Accept-Encoding
ETag: "62176d0c-90d"
Accept-Ranges: bytes

Please use http://arduino.tips/asciilogo.txt via HTTP

           `:;;;,`                      .:;;:.           
        .;;;;;;;;;;;`                :;;;;;;;;;;:     TM 
      `;;;;;;;;;;;;;;;`            :;;;;;;;;;;;;;;;      
     :;;;;;;;;;;;;;;;;;;         `;;;;;;;;;;;;;;;;;;     
    ;;;;;;;;;;;;;;;;;;;;;       .;;;;;;;;;;;;;;;;;;;;    
   ;;;;;;;;:`   `;;;;;;;;;     ,;;;;;;;;.`   .;;;;;;;;   
  .;;;;;;,         :;;;;;;;   .;;;;;;;          ;;;;;;;  
  ;;;;;;             ;;;;;;;  ;;;;;;,            ;;;;;;. 
 ,;;;;;               ;;;;;;.;;;;;;`              ;;;;;; 
 ;;;;;.                ;;;;;;;;;;;`      ```       ;;;;;`
 ;;;;;                  ;;;;;;;;;,       ;;;       .;;;;;
`;;;;:                  `;;;;;;;;        ;;;        ;;;;;
,;;;;`    `,,,,,,,,      ;;;;;;;      .,,;;;,,,     ;;;;;
:;;;;`    .;;;;;;;;       ;;;;;,      :;;;;;;;;     ;;;;;
:;;;;`    .;;;;;;;;      `;;;;;;      :;;;;;;;;     ;;;;;
.;;;;.                   ;;;;;;;.        ;;;        ;;;;;
 ;;;;;                  ;;;;;;;;;        ;;;        ;;;;;
 ;;;;;                 .;;;;;;;;;;       ;;;       ;;;;;,
 ;;;;;;               `;;;;;;;;;;;;                ;;;;; 
 `;;;;;,             .;;;;;; ;;;;;;;              ;;;;;; 
  ;;;;;;:           :;;;;;;.  ;;;;;;;            ;;;;;;  
   ;;;;;;;`       .;;;;;;;,    ;;;;;;;;        ;;;;;;;:  
    ;;;;;;;;;:,:;;;;;;;;;:      ;;;;;;;;;;:,;;;;;;;;;;   
    `;;;;;;;;;;;;;;;;;;;.        ;;;;;;;;;;;;;;;;;;;;    
      ;;;;;;;;;;;;;;;;;           :;;;;;;;;;;;;;;;;:     
       ,;;;;;;;;;;;;;,              ;;;;;;;;;;;;;;       
         .;;;;;;;;;`                  ,;;;;;;;;:         
                                                         
                                                         
                                                         
                                                         
    ;;;   ;;;;;`  ;;;;:  .;;  ;; ,;;;;;, ;;. `;,  ;;;;   
    ;;;   ;;:;;;  ;;;;;; .;;  ;; ,;;;;;: ;;; `;, ;;;:;;  
   ,;:;   ;;  ;;  ;;  ;; .;;  ;;   ,;,   ;;;,`;, ;;  ;;  
   ;; ;:  ;;  ;;  ;;  ;; .;;  ;;   ,;,   ;;;;`;, ;;  ;;. 
   ;: ;;  ;;;;;:  ;;  ;; .;;  ;;   ,;,   ;;`;;;, ;;  ;;` 
  ,;;;;;  ;;`;;   ;;  ;; .;;  ;;   ,;,   ;; ;;;, ;;  ;;  
  ;;  ,;, ;; .;;  ;;;;;:  ;;;;;: ,;;;;;: ;;  ;;, ;;;;;;  
  ;;   ;; ;;  ;;` ;;;;.   `;;;:  ,;;;;;, ;;  ;;,  ;;;;   

Disconnecting.
Received 2574 bytes in 0.3492, rate = 7.37 kbytes/second
```

4. If we change to use the correct IPAddress of `arduino.cc`, which is IPAddress server(100,24,172,113), we can't connect

- code
- 
```
//char server[] = "arduino.cc";  // also change the Host line in httpRequest()
IPAddress server(100,24,172,113);
```

- Terminal 

```
Starting WebClient using QNEthernet Library
Initialize Ethernet using static IP => IP Address = 192.168.2.222
Connecting to 100.24.172.113...
Connection failed

Disconnecting.
Received 0 bytes in 0.0000, rate = nan kbytes/second
```

khoih-prog, 2022-03-14:
### More tests on yahoo.com

#### 1. ping

```
$ ping yahoo.com
PING yahoo.com (98.137.11.163) 56(84) bytes of data.
64 bytes from yahoo.com (98.137.11.163): icmp_seq=1 ttl=52 time=64.9 ms
64 bytes from yahoo.com (98.137.11.163): icmp_seq=2 ttl=52 time=64.9 ms
64 bytes from yahoo.com (98.137.11.163): icmp_seq=3 ttl=52 time=64.8 ms
64 bytes from yahoo.com (98.137.11.163): icmp_seq=4 ttl=52 time=64.9 ms
```

#### 2. Using hostname `yahoo.com` => OK

```
Starting WebClient using QNEthernet Library
Initialize Ethernet using static IP => IP Address = 192.168.2.222
Connecting to yahoo.com...
Connected to 98.137.11.164
HTTP/1.1 404 Not Found on Accelerator
Date: Mon, 14 Mar 2022 03:44:09 GMT
Connection: close
Server: ATS
Cache-Control: no-store
Content-Type: text/html
Content-Language: en
X-Frame-Options: SAMEORIGIN
Content-Length: 4792

<!DOCTYPE html>
<html lang="en-us">
  <head>
    <meta http-equiv="content-type" content="text/html; charset=UTF-8">
    <meta charset="utf-8">
    <title>Yahoo</title>
    <meta name="viewport" content="width=device-width,initial-scale=1,minimal-ui">
    <meta http-equiv="X-UA-Compatible" content="IE=edge,chrome=1">
    <style>
      html {
          height: 100%;
      }
      body {
          background: #fafafc url(https://s.yimg.com/nn/img/sad-panda-201402200631.png) 50% 50%;
          background-size: cover;
          height: 100%;
          text-align: center;
          font: 300 18px "helvetica neue", helvetica, verdana, tahoma, arial, sans-serif;
          margin: 0;
      }
      table {
          height: 100%;
          width: 100%;
          table-layout: fixed;
          border-collapse: collapse;
          border-spacing: 0;
          border: none;
      }
      h1 {
          font-size: 42px;
          font-weight: 400;
          color: #400090;
      }
      p {
          color: #1A1A1A;
      }
      #message-1 {
          font-weight: bold;
          margin: 0;
      }
[... 106 lines cut ...]
```

#### 3. Using IPAddress server(98,137,11,163) for `yahoo.com` => not OK

```
Starting WebClient using QNEthernet Library
Initialize Ethernet using static IP => IP Address = 192.168.2.222
Connecting to 98.137.11.163...
Connection failed

Disconnecting.
Received 0 bytes in 0.0000, rate = nan kbytes/second
```

#### 4. Using IPAddress server(98,137,11,164) for `yahoo.com` as in terminal 2 => still not OK

```
Starting WebClient using QNEthernet Library
Initialize Ethernet using static IP => IP Address = 192.168.2.222
Connecting to 98.137.11.164...       <===== Using same IP as in #### 2
Connection failed

Disconnecting.
Received 0 bytes in 0.0000, rate = nan kbytes/second
```

ssilverman, 2022-03-14:
Thanks for the details. I will have a look.

khoih-prog, 2022-03-14:
The error is in function

```
err_t tcp_connect(struct tcp_pcb *pcb, const ip_addr_t *ipaddr, u16_t port, tcp_connected_fn connected)
```

with the return value `ERR_RTE = -4` => no route ??

ssilverman, 2022-03-14:
What are you setting the IP/netmask/gateway to your n the above tests?

khoih-prog, 2022-03-14:
They are set according to my local network, and they are certainly correct as they are OK for `staticIP + hostname`

https://github.com/khoih-prog/QNEthernet/blob/master/examples/WebClient/WebClient.ino#L28-L34
https://github.com/khoih-prog/QNEthernet/blob/master/examples/WebClient/WebClient.ino#L54-L63

```
#if !USING_DHCP
  // Set the static IP address to use if the DHCP fails to assign
  IPAddress myIP(192, 168, 2, 222);
  IPAddress myNetmask(255, 255, 255, 0);
  IPAddress myGW(192, 168, 2, 1);
  IPAddress mydnsServer(8, 8, 8, 8);
#endif

#if USING_DHCP
  // Start the Ethernet connection, using DHCP
  Serial.print("Initialize Ethernet using DHCP => ");
  Ethernet.begin();
#else   
  // Start the Ethernet connection, using static IP
  Serial.print("Initialize Ethernet using static IP => ");
  Ethernet.begin(myIP, myNetmask, myGW);
  Ethernet.setDNSServerIP(mydnsServer);
#endif
```

ssilverman, 2022-03-14:
I think I figured it out; there appear to be two factors:

1. The "route error" (ERR_RTE=-4) is because the link isn't yet detected to be up when a connection is attempted. To solve this, after setting a static IP, you can call a new `Ethernet.waitForLink(timeout)` function that waits for the link to be detected. Alternatively, you can check if there's a link each time you attempt to connect as a client. I don't have a preference for either of the two approaches; it will depend on how the code is structured or intended to be used.

2. It's possible that the default connection timeout, 1000ms, isn't long enough. Try setting this higher. See: [url]https://www.arduino.cc/en/Reference/EthernetClientSetConnectionTimeout[/url]

khoih-prog, 2022-03-14:
> To solve this, after setting a static IP, you can call a new Ethernet.waitForLink(timeout) function that waits for the link to be detected.

But why the staticIP is working normally when using server's `hostname`, which takes more time to convert to serverIP ?

in the [**WebClient**](https://github.com/khoih-prog/QNEthernet/tree/master/examples/WebClient)  code, there is already 5000ms timeout waiting for localIP

```
  if (!Ethernet.waitForLocalIP(5000))
  {
    Serial.println("Failed to configure Ethernet");

    if (!Ethernet.linkStatus())
    {
      Serial.println("Ethernet cable is not connected.");
    }

    // Stay here forever
    while (true)
    {
      delay(1);
    }
  }
  else
  {
    Serial.print("IP Address = ");
    Serial.println(Ethernet.localIP());
  }
```

Moreover, there is no `waitForLink()` function anywhere in your lib.

I guess this is very elusive bug, don't know there's anything related to the T4.1 Ethernet chip, because `NativeEthernet` lib has the exactly same `staticIP + serverIP` issue.

khoih-prog, 2022-03-14:
OMG, you're right.

If I change https://github.com/khoih-prog/QNEthernet/blob/master/examples/WebClient/WebClient.ino#L87 from

```
delay(1000);
```

to

```
#if USING_DHCP
  delay(1000);
#else  
  delay(2000);
#endif
```

then it's OK with `staticIP + serverIP`

```

Starting WebClient using QNEthernet Library
Initialize Ethernet using static IP => IP Address = 192.168.2.222
ipaddr = 3724716224
Connecting to 74.6.143.25...
QNEthernetClient::connect(IPAddress): ip = 74.6.143.25, port = 80
QNEthernetClient::connect(IPAddress): ipaddr = 74.6.143.25
QNEthernetClient::connect(ip_addr_t): enter
QNEthernetClient::connect(ip_addr_t): ip_addr_t = 74.6.143.25
ConnectionManager::connect(ip_addr_t): ipaddr = 74.6.143.25, port = 80
ConnectionManager::connect(ip_addr_t): OK tcp_new
ConnectionManager::connect(ip_addr_t): OK tcp_bind
ConnectionManager::connect(ip_addr_t): pcb->local_ip = 0.0.0.0
ConnectionManager::connect(ip_addr_t): error tcp_connect, tcp_connect() = 0
ConnectionManager::connect(ip_addr_t): OK tcp_connect
ConnectionManager::connect(ip_addr_t): OK, pcb->local_ip = 192.168.2.222
QNEthernetClient::connect(ip_addr_t): OK
Connected to 74.6.143.25
HTTP/1.1 404 Not Found on Accelerator
Date: Mon, 14 Mar 2022 18:09:29 GMT
Connection: close
Server: ATS
Cache-Control: no-store
Content-Type: text/html
Content-Language: en
X-Frame-Options: SAMEORIGIN
Content-Length: 4792

<!DOCTYPE html>
<html lang="en-us">
  <head>
    <meta http-equiv="content-type" content="text/html; charset=UTF-8">
    <meta charset="utf-8">
    <title>Yahoo</title>
    <meta name="viewport" content="width=device-width,initial-scale=1,minimal-ui">
    <meta http-equiv="X-UA-Compatible" content="IE=edge,chrome=1">
    <style>
      html {
          height: 100%;
      }
      body {
          background: #fafafc url(https://s.yimg.com/nn/img/sad-panda-201402200631.png) 50% 50%;
          background-size: cover;
          height: 100%;
          text-align: center;
          font: 300 18px "helvetica neue", helvetica, verdana, tahoma, arial, sans-serif;
          margin: 0;
      }
      table {
[... 102 lines cut ...]
    generateBRBMarkup(hostname, params);

  </script>
  <noscript>
  <table>
    <tbody>
      <tr>
        <td>
          <div id="englishContent">
            <h1 style="margin-top:20px;">Will be right back...</h1>
            <p id="message-1">Thank you for your patience.</p>
            <p id="message-2">Our engineers are working quickly to resolve the issue.</p>
          </div>
        </td>
      </tr>
    </tbody>
  </table>
  </noscript>
  </body>
</html>

Disconnecting.
Received 5025 bytes in 0.2518, rate = 19.95 kbytes/second

```

ssilverman, 2022-03-14:
I just pushed some new commits; I hadn't pushed it yet. :)

`hostname` question: My guess is that it has something to do with the fact that DNS lookups are UDP and there are retries by the time a link has been established, whereas TCP connections require some back-and-forth communication.

5000ms timeout waiting for localIP question: If a static IP is set, then there is immediately a valid localIP; no link is required. WIth DHCP, if it receives an IP address, a link has already been established because otherwise no traffic would come in.

closed 2022-03-14 by khoih-prog

ssilverman, 2022-03-14:
I don't want to make functions blocking without a specified timeout. What are your thoughts about `begin(staticIP-stuff)` followed by `Ethernet.waitForLink(timeout)`? I feel that the "auto-wait" you suggest is better suited as a user function that wraps these two calls. I feel there's already too many `begin` functions.

Additionally, I'm not a fan of setting some easy-to-forget state that affects the behaviour of other functions. The necessary building blocks are already there and it's my preference to provide the building blocks but not all possible variations of options.

The best way to do all this, in my opinion, is to use the listener approach and watch for events, and then take actions on those events. The approach where all the state is detected once at the beginning of the program and then never again is brittle.

khoih-prog, 2022-03-14:
> The best way to do all this instead, in my opinion, is to use the listener approach and watch for events, and then take actions on those events. The approach where all the state is detected once at the beginning of the program and then never again is brittle.

I agree with this approach, and this can only be decided by checking some ready events in the QNEthernet library. Certainly it's all up to your decision, which I trust is the good one.

khoih-prog, 2022-03-14:
I will update all the QNEthernet and NativeEthernet-related code and publish a new release

khoih-prog, 2022-03-14:
I prefer 1., certainly with these preferences

1. Follow some standard API, if exists, to make code more portable
2. Clean API
3. Smart-enough API to fulfill complex tasks normal users can't have time or knowledge to study and understand. That's why libraries are necessary step-stones for complex projects.
For example, this issue can be so elusive and almost unsolvable, until you, as the author, have the correct direction.

Anyway, I spent some time going thru your lib last night, and still think yours is still one of the best libs I've known. I can trust everything you do now.

khoih-prog, 2022-03-14:
Hi @ssilverman 

The new [**WebSockets2_Generic releases v1.10.2**](https://github.com/khoih-prog/WebSockets2_Generic/releases/tag/v1.10.2) has just been published. 

Your contribution, bug-fixing suggestion is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/WebSockets2_Generic#contributions-and-thanks)

---

### Release v1.10.2

1. Fix bug when using `QNEthernet` staticIP. Check [QNEthernet and NativeEthernet staticIP not working with WS Server #39](https://github.com/khoih-prog/WebSockets2_Generic/issues/39)
2. Add staticIP option to `NativeEthernet` examples
2. Update `Packages' Patches`