khoih-prog/WebSockets2_Generic #41: what is websocketclient if no incoming request?

sdetweil, 2022-03-28:
after
```
client = SocketsServer.accept();
```
after no incoming client connection request,

is it a single client object? only one client allowed at a time?

how can I push messages to the client asynchronously?

I would build an array of clients from accept
and loop thru each for incoming messages,
but send based on other events, data status change, elasped time....

khoih-prog, 2022-03-29:
As this library based on the [**ArduinoWebsockets** Library](https://github.com/gilmaimon/ArduinoWebsockets), just ported to many more boards besides the popular `ESP32/ESP8266`, I suggest that you test on the `ESP32/ESP8266 `to see if there is the same issue / question. 

If yes, please post the issue / question on the [**ArduinoWebsockets** Library](https://github.com/gilmaimon/ArduinoWebsockets) as I don't have time now to spend and deal / solve the `basic original design` of the library.

I also suggest that you have to include the [**Minimal, Reproducible Example**](https://stackoverflow.com/help/minimal-reproducible-example), as no one can have the time/patience to `read-the-tea-leaves` to understand your issue and and recreate the whole test case.

We'd appreciate if you post the result / link here for we all to know the answer / solution.

Anyway, I'm closing this issue here as this is not a proven bug of the library yet and won't spend time to work on it.

Good Luck,

closed 2022-03-29 by khoih-prog

sdetweil, 2022-03-29:
thanks. this was an info request. an issue type you dont provide.