khoih-prog/WebSockets2_Generic #42: ESP32-Server Not Working Properly

ashutosh7777, 2022-04-06:
i have uploaded esp32 server code from examples.
[image]

This is my program
/****************************************************************************************************************************
  ESP32-Server.ino
  For ESP32.

  Based on and modified from Gil Maimon's ArduinoWebsockets library https://github.com/gilmaimon/ArduinoWebsockets
  to support STM32F/L/H/G/WB/MP1, nRF52 and SAMD21/SAMD51 boards besides ESP8266 and ESP32

  The library provides simple and easy interface for websockets (Client and Server).

  Built by Khoi Hoang https://github.com/khoih-prog/Websockets2_Generic
  Licensed under MIT license
 *****************************************************************************************************************************/
/****************************************************************************************************************************
  ESP32 Websockets Server : Minimal ESP32 Websockets Server

  This sketch:
        1. Connects to a WiFi network
        2. Starts a websocket server on port 8080
        3. Waits for connections
        4. Once a client connects, it wait for a message from the client
        5. Sends an "echo" message to the client
        6. closes the connection and goes back to step 3

  Hardware:
        For this sketch you only need an ESP32 board.

  Originally Created  : 15/02/2019
  Original Author     : By Gil Maimon
  Original Repository : https://github.com/gilmaimon/ArduinoWebsockets

*****************************************************************************************************************************/

#include "defines.h"

#define DEBUG_LOCAL   2

#include <WebSockets2_Generic.h>
#include <WiFi.h>

using namespace websockets2_generic;

WebsocketsServer server;

void heartBeatPrint(void)
{
  static int num = 1;

  if (WiFi.status() == WL_CONNECTED)
    Serial.print("H");        // H means server WiFi connected
  else
    Serial.print("F");        // F means server WiFi not connected

  if (num == 80)
  {
    Serial.println();
    num = 1;
  }
  else if (num++ % 10 == 0)
  {
    Serial.print(" ");
  }
}

void check_status()
{
  static unsigned long checkstatus_timeout = 0;

  //KH
#define HEARTBEAT_INTERVAL    10000L
  // Print hearbeat every HEARTBEAT_INTERVAL (10) seconds.
  if ((millis() > checkstatus_timeout) || (checkstatus_timeout == 0))
  {
    heartBeatPrint();
    checkstatus_timeout = millis() + HEARTBEAT_INTERVAL;
  }
}

void setup()
{
  Serial.begin(115200);
  while (!Serial);

  Serial.print("\nStart ESP32-Server on ");
  Serial.println(ARDUINO_BOARD);
  Serial.println(WEBSOCKETS2_GENERIC_VERSION);

  //  WiFi.mode(WIFI_STA);

  //  WiFi.config(serverIP, static_GW, static_SN);

  // Connect to wifi
  WiFi.begin(ssid, password);

  // Wait some time to connect to wifi
  for (int i = 0; i < 15 && WiFi.status() != WL_CONNECTED; i++)
  {
    Serial.print(".");
    delay(1000);
  }

  if (WiFi.status() == WL_CONNECTED)
    Serial.println("\nWiFi connected");
  else
  {
    Serial.println("\nNo WiFi");
    return;
  }

  Serial.print(server.available() ? "WebSockets Server Running and Ready on " : "Server Not Running on ");
  server.listen(WEBSOCKETS_PORT);
  Serial.println(BOARD_NAME);
  Serial.print("IP address: ");
  Serial.print(WiFi.localIP());     //You can get IP address assigned to SAMD
  Serial.print(", Port: ");
  Serial.println(WEBSOCKETS_PORT);    // Websockets Server Port
}

void loop()
{
   check_status();

  WebsocketsClient client = server.accept();

    if (client.available())
    {
  WebsocketsMessage msg = client.readNonBlocking();
  WebsocketsMessage msg = client.readBlocking();

  // log
  Serial.print("Got Message: ");
  Serial.println(msg.data());

  // return echo
      client.send("Echo: " + msg.data());

  // close the connection
  //    client.close();
  }
  client.ping();
  //delay(1000);
}

**PROBLEM IS**
Client gets connected to server but after delivering 1 msg it gets disconnected. i tgried another libraries client code works properly. while code keeps me disconnecting and doesnt receive any other msg.

[image]

any help will be great for us.
thank you in advance

khoih-prog, 2022-04-07:
As you're using **ESP32**, it's better to test using the original [**ArduinoWebsockets** library](https://github.com/gilmaimon/ArduinoWebsockets), this library is based upon. 

If there is problem, it's better to post the issue on that [**ArduinoWebsockets** library](https://github.com/gilmaimon/ArduinoWebsockets).

I'm closing this issue now and appreciate if you can post the link to your new issue.

Good Luck,

closed 2022-04-07 by khoih-prog

khoih-prog, 2022-04-08:
For the record, answered in [**ArduinoWebsockets**](https://github.com/gilmaimon/ArduinoWebsockets)

> When the client object goes out of scope, the connection is closed. If you want to keep receiving/sending messages, you should not let the object go out of scope. Instead keep interacting with it until you want it closed. You could check the wiki for more information and advanced examples for handling multiple clients and so on.