khoih-prog/WebSockets2_Generic #53: Uploading and running example of ESP "Minimal-Esp32-Client" error [  4159][E][WiFiClient.cpp:313] setSocketOption(): fail on -1, errno: 9, "Bad file number"

yoavshah, 2022-10-19:
### Describe the bug

I tried to run the example of ESP32 called "Minimal-Esp32-Client" (and other examples, but this one is the easiest to reproduce), then the monitor printed 

[  4159][E][WiFiClient.cpp:313] setSocketOption(): fail on -1, errno: 9, "Bad file number"

and then 
Not Connected!

### Steps to Reproduce

I am using visual studio code with PlatformIO and created a new project of ESP32 on Arduino framework.
Just put the example code "Minimal-Esp32-Client" with your wifi ssid and password and the websocket server to connect to and it will do it

PlatformIO version: Core 6.1.4 - Home 3.4.3
Core Version: I am not sure, i think Arduino-mbed
Board type: ESP32
Contextual information: I tried to run the example code
Using Windows 10

[image]

yoavshah, 2022-10-19:
When running in ArduinoIDE I still got the error message, but I could print more debug line
[image]

khoih-prog, 2022-10-20:
Do some basic research, test some more examples before posting any issue.

I'm closing the issue unless you prove this is a bug of this library.

As you're using ESP32, I suggest you post any issue next time in [**ArduinoWebsockets** library](https://github.com/gilmaimon/ArduinoWebsockets)

closed 2022-10-20 by khoih-prog

khoih-prog, 2022-10-20:
```
Start Minimal-ESP32-Client on ESP32_DEV
WebSockets2_Generic v1.13.0
..Connected to Wifi, Connecting to WebSockets Server @192.168.2.30
[WS] WebsocketsClient::generateHandshake: base64Authorization = 
[WS] WebsocketsClient::generateHandshake: handshake = GET / HTTP/1.1
Host: 192.168.2.30
Sec-WebSocket-Key: MDEyMzQ1Njc4OWFiY2RlZg==
Upgrade: websocket
Connection: Upgrade
Sec-WebSocket-Version: 13
User-Agent: TinyWebsockets Client
Authorization: Basic 
Origin: https://github.com/khoih-prog/Websockets2_Generic

[WS] WebsocketsClient::connect: base64Authorization = 
[WS] WebsocketsClient::generateHandshake: key = Upgrade
[WS] WebsocketsClient::generateHandshake: value = websocket
[WS] WebsocketsClient::generateHandshake: key = Connection
[WS] WebsocketsClient::generateHandshake: value = Upgrade
[WS] WebsocketsClient::generateHandshake: key = Sec-WebSocket-Accept
[WS] WebsocketsClient::generateHandshake: value = BACScCJPNqyz+UBoqMH89VmURoA=
[WS] WebsocketsClient::generateHandshake: key = Origin
[WS] WebsocketsClient::generateHandshake: value = https://github.com/khoih-prog/Websockets2_Generic
Connnection Opened
Connected!
Got Message: Hello to Server from ESP32_DEV
Got a Pong!

```