khoih-prog/WebSockets2_Generic #63: Examples Generic WiFiNINA RP2040 ServerAllFunctionsDemo does not work

jnanneng2, 2023-01-16:
### Describe the bug

This sample is non functional:  https://github.com/khoih-prog/WebSockets2_Generic/tree/master/examples/Generic/WiFiNINA/RP2040/RP2040_ServerAllFunctionsDemo

### Steps to Reproduce

Using an Arudino Nano RP2040 Connection, upload the noted example. 

### Expected behavior

The RGB sliders would actually work, changing the onboard LED colors.
Sending data to the hosted socket would also display the data received.

### Actual behavior

The RGB sliders do not do anything.  The onboard LED remains off.
When sending data to the hosted socket, the board acknowledges that a message has been received, however zero data is displayed, even when sending data to it.

### Information

Arduino IDE version: 2.0.3
Arduino Mbed OS RP2040 3.5.1
NANO RP2040 CONNECT

khoih-prog, 2023-01-16:
The example doesn't actually control the RGB LED, just a simple demo for WebSocket server.

To actually control the LEDs, you have to adapt the code from

https://github.com/Links2004/arduinoWebSockets/blob/323592f622e0ec8f9ce1f995c5777d9bbaaae1ec/examples/esp8266/WebSocketServerAllFunctionsDemo/WebSocketServerAllFunctionsDemo.ino#L28-L59

to receive command ( in `msg.data()` ) and contriol the LEDs yourself

```
void webSocketEvent(uint8_t num, WStype_t type, uint8_t * payload, size_t length) 
{
  switch(type) 
  {
    case WStype_DISCONNECTED:
    
      Serial.printf("[%u] Disconnected!\n", num);
      
      break;
    
    case WStype_CONNECTED: 
      {
        IPAddress ip = webSocket.remoteIP(num);
        Serial.printf("[%u] Connected from %d.%d.%d.%d url: %s\n", num, ip[0], ip[1], ip[2], ip[3], payload);

        // send message to client
        webSocket.sendTXT(num, "Connected");
      }
  
      break;
  
  case WStype_TEXT:
    Serial.printf("[%u] get Text: %s\n", num, payload);

    if(payload[0] == '#') 
    {
      // we get RGB data

      // decode rgb data
      uint32_t rgb = (uint32_t) strtol((const char *) &payload[1], NULL, 16);

      analogWrite(LED_RED, ((rgb >> 16) & 0xFF));
      analogWrite(LED_GREEN, ((rgb >> 8) & 0xFF));
      analogWrite(LED_BLUE, ((rgb >> 0) & 0xFF));
    }

    break;
  }
}
```

As I have no intention to do this, please use it **as is** or **ignore** the example

It maybe better to delete the example in the near future to avoid confusion.

Good Luck,

closed 2023-01-16 by khoih-prog

jnanneng2, 2023-01-16:
@khoih-prog 

Do you have any working example of duplex socket communication using this library and a NANO RP2040 CONNECT?  Everything I've tried does not work.  The examples also seem to not work.  The examples you linked to do not use this library, of which there are great differences in the implementation.  I cannot find API documentation on this library (does that exist somewhere).  

Thus, kind of stuck; looking for some guidance as to how to get a duplex socket up and running with this library; with the server socket being hosed on the RP2040.

khoih-prog, 2023-01-16:
Try to switch to  cheaper and faster [**RP2040W**](https://github.com/khoih-prog/AsyncWebServer_RP2040W#currently-supported-boards), using [**AsyncWebServer_RP2040W**](https://github.com/khoih-prog/AsyncWebServer_RP2040W) with [**arduino-pico**](https://github.com/earlephilhower/arduino-pico) core

MirkoGelsomini, 2023-01-21:
> @khoih-prog
> 
> Do you have any working example of duplex socket communication using this library and a NANO RP2040 CONNECT? Everything I've tried does not work. The examples also seem to not work. The examples you linked to do not use this library, of which there are great differences in the implementation. I cannot find API documentation on this library (does that exist somewhere)?
> 
> Thus, kind of stuck; looking for some guidance as to how to get a duplex socket up and running with this library; with the server socket being hosted on the RP2040.

I agree with @jnanneng2 , the code you are referring to seems to come from a different repo using a different library.
Can you share a full working example?