khoih-prog/WebSockets2_Generic #64: RP2040 WebSocket immediate disconnection

MirkoGelsomini, 2023-01-21:
### Bug

Starting from "RP2040_ServerAllFunctionsDemo.ino" from  File > Examples > WebSockets2_Generic > Generic > WiFiNINA > RP2040 .
After some fixes (explained below) I can connect but I cannot maintain the communication open.

### Steps to Reproduce

- Used generic RP2040_ServerAllFunctionsDemo.ino  from  File > Examples > WebSockets2_Generic > Generic > WiFiNINA > RP2040 .
- Set correct ssid and password in defines.h
- Set the correct wsServer url in line 100
- Went to the ip address:80 and saw the RGB sliders

### Expected behavior

Socket connection opened and communication happening (not interested in RGB to really work but the communication to being established).

### Actual behavior

1. ***************************************************************************************

PROBLEM: Javascript error: "Uncaught ReferenceError: wsServer is not defined" . 

SOLUTION: Solved by replacing in line 106 this:
`var connection = new WebSocket(wsServer, ['arduino']);`
with:
`var connection = new WebSocket('ws://'+window.location.hostname+':8080/', ['arduino']);`
Compiled and uploaded new code.

RESULT: problem 1 solved, but I receive "WebSocket Error"

2.  ***************************************************************************************

PROBLEM: WebSocket Error

SOLUTION: Solved by developing my own script available below:

        
        var wsClientOutput;

        function init() {
            testWebSocket();
        }

        function testWebSocket() {
            websocket = new WebSocket("ws://192.168.2.123:8080/"); //change with yours
            websocket.onopen = function (evt) {
                onOpen(evt)
            };
            websocket.onclose = function (evt) {
                onClose(evt)
            };
            websocket.onmessage = function (evt) {
                onMessage(evt)
            };
            websocket.onerror = function (evt) {
                onError(evt)
            };
        }

        function onOpen(evt) {
           console.log("OPEN")
        }

        function onClose(evt) {
            console.log("DISCONNECTED");
        }

        function onMessage(evt) {
            console.log('RESPONSE: ' + evt.data);
        }

        function onError(evt) {
            console.log('ERROR:' + evt.data);
        }

        function doSend(message) {
            console.log("SENT: " + message);
            websocket.send(message);
        }

       
        window.addEventListener("load", init, false);

    
RESULT: I can connect but I quickly disconnect.

3.  ***************************************************************************************

PROBLEM: Immediate disconnection after connection

SOLUTION:  I removed line 140 and line 143 from "RP2040_ServerAllFunctionsDemo.ino "
Compiled and uploaded new code.

RESULT: same error. Immediate disconnection after connection.

### Information

Please ensure to specify the following, or your post will be ignored and deleted:

* Arduino IDE version: 2.0.3
* `RP2040` Core Version: Arduino-mbed OS Nano 3.5.4
* `RP2040` Board type: NANO_RP2040_CONNECT
* Contextual information: maintain connection without automatic disconnection and stream some sensors values.
* Operating system (Windows, Ubuntu, etc.): Windows 10
* Network: Wi-Fi

khoih-prog, 2023-01-25:
Hi @MirkoGelsomini 

> PROBLEM: Immediate disconnection after connection

That's the way the example is written and you have to read and understand to use

https://github.com/khoih-prog/WebSockets2_Generic/blob/aa30bc03f604efe07d6c08214391a5ad98676776/examples/Generic/RP2040W/RP2040W_ServerAllFunctionsDemo/RP2040W_ServerAllFunctionsDemo.ino#L153-L157

**You must modify the example depending on your different use-case.**

You have to write the correct and matching WS Client to reconnect in `loop` to use with this example.

---

For example, the WS server is @ `192.168.2.118:8080`

### Python WS Client code `WSClient.py`

```
# Run by $ python3.8 WSClient.py
# From websocket-client package. Install by $ pip3 install websocket-client
import websocket
import time

ws = websocket.WebSocket()

while True:
    ws.connect("ws://192.168.2.118:8080")
    ws.send("Hello, Server")
    result = ws.recv()
    print(result)
    time.sleep(10)
```

### WS Client Terminal

```
kh@kh-Inspiron-3593:~/Downloads/Nodejs/WSClient_Python$ python3.8 WSClient.py
Echo: 
Echo: 
Echo: 
Echo: 
Echo: 
Echo: 
...
```

---

### Debug Terminal for WS Server

```
Starting RP2040W_ServerAllFunctionsDemo on RASPBERRY_PI_PICO_W
WebSockets2_Generic v1.13.2
Connecting to SSID: HueNet
SSID: HueNet
Local IP Address: 192.168.2.118
WebSockets Server Running and Ready on RASPBERRY_PI_PICO_W
IP address: 192.168.2.118, Port: 8080
wsServer: ws://192.168.2.95:8080/
[WS] WebsocketsServer::recvHandshakeRequest: key = Upgrade
[WS] WebsocketsServer::recvHandshakeRequest: value = websocket
[WS] WebsocketsServer::recvHandshakeRequest: lowkey = upgrade
[WS] WebsocketsServer::recvHandshakeRequest: lowvalue = websocket
[WS] WebsocketsServer::recvHandshakeRequest: key = Host
[WS] WebsocketsServer::recvHandshakeRequest: value = 192.168.2.118:8080
[WS] WebsocketsServer::recvHandshakeRequest: lowkey = host
[WS] WebsocketsServer::recvHandshakeRequest: lowvalue = 192.168.2.118:8080
[WS] WebsocketsServer::recvHandshakeRequest: key = Origin
[WS] WebsocketsServer::recvHandshakeRequest: value = http://192.168.2.118:8080
[WS] WebsocketsServer::recvHandshakeRequest: lowkey = origin
[WS] WebsocketsServer::recvHandshakeRequest: lowvalue = http://192.168.2.118:8080
[WS] WebsocketsServer::recvHandshakeRequest: key = Sec-WebSocket-Key
[WS] WebsocketsServer::recvHandshakeRequest: value = kPyLjt0Z8UfeSFy0purr3Q==
[WS] WebsocketsServer::recvHandshakeRequest: lowkey = sec-websocket-key
[WS] WebsocketsServer::recvHandshakeRequest: lowvalue = kPyLjt0Z8UfeSFy0purr3Q==
[WS] WebsocketsServer::recvHandshakeRequest: key = Sec-WebSocket-Version
[WS] WebsocketsServer::recvHandshakeRequest: value = 13
[WS] WebsocketsServer::recvHandshakeRequest: lowkey = sec-websocket-version
[WS] WebsocketsServer::recvHandshakeRequest: lowvalue = 13
[WS] WebsocketsServer::recvHandshakeRequest: key = Connection
[WS] WebsocketsServer::recvHandshakeRequest: value = Upgrade
[WS] WebsocketsServer::recvHandshakeRequest: lowkey = connection
[WS] WebsocketsServer::recvHandshakeRequest: lowvalue = upgrade
Got Message: 
[WS] WebsocketsServer::recvHandshakeRequest: key = Upgrade
[WS] WebsocketsServer::recvHandshakeRequest: value = websocket
[WS] WebsocketsServer::recvHandshakeRequest: lowkey = upgrade
[WS] WebsocketsServer::recvHandshakeRequest: lowvalue = websocket
[WS] WebsocketsServer::recvHandshakeRequest: key = Host
[WS] WebsocketsServer::recvHandshakeRequest: value = 192.168.2.118:8080
[WS] WebsocketsServer::recvHandshakeRequest: lowkey = host
[WS] WebsocketsServer::recvHandshakeRequest: lowvalue = 192.168.2.118:8080
[WS] WebsocketsServer::recvHandshakeRequest: key = Origin
[WS] WebsocketsServer::recvHandshakeRequest: value = http://192.168.2.118:8080
[WS] WebsocketsServer::recvHandshakeRequest: lowkey = origin
[WS] WebsocketsServer::recvHandshakeRequest: lowvalue = http://192.168.2.118:8080
[WS] WebsocketsServer::recvHandshakeRequest: key = Sec-WebSocket-Key
[WS] WebsocketsServer::recvHandshakeRequest: value = nrATeKoxvo2o9DEoFdiTcg==
[WS] WebsocketsServer::recvHandshakeRequest: lowkey = sec-websocket-key
[WS] WebsocketsServer::recvHandshakeRequest: lowvalue = nrATeKoxvo2o9DEoFdiTcg==
[WS] WebsocketsServer::recvHandshakeRequest: key = Sec-WebSocket-Version
[WS] WebsocketsServer::recvHandshakeRequest: value = 13
[WS] WebsocketsServer::recvHandshakeRequest: lowkey = sec-websocket-version
[WS] WebsocketsServer::recvHandshakeRequest: lowvalue = 13
[WS] WebsocketsServer::recvHandshakeRequest: key = Connection
[WS] WebsocketsServer::recvHandshakeRequest: value = Upgrade
[WS] WebsocketsServer::recvHandshakeRequest: lowkey = connection
[WS] WebsocketsServer::recvHandshakeRequest: lowvalue = upgrade
Got Message: 
...
```

closed 2023-01-25 by khoih-prog

khoih-prog, 2023-01-25:
FYI, I just add [**WSClient_Python**](https://github.com/khoih-prog/WebSockets2_Generic/tree/master/examples/WSClient_Python) example to be used with the WS Server examples